betaflight/betaflight #4966: [Feature Request] Total distance flown on post-flight OSD statistics screen

adamtfc87, 2018-01-15:
I was wondering if it's possible to get total distance flown displayed on the after flight stats screen like iNav has. There's a lot of people flying long range with betaflight now and I think it would be a great addition. Also, is it possible to display gps speed in miles per hour? Thanks!

DanNixon, 2018-01-15:
Could you open a second issue for the speed units.

adamtfc87, 2018-01-15:
Just put it into a separate request. Would adding total distance flown be a very difficult thing?

adamtfc87, 2018-01-22:
So... is this even a possibility? Seems not too many people care about this request.. but I think it would be such a nice feature.

mikeller, 2018-01-23:
@adamtfc87: You opened this one week ago. I don't know if you've got a realistic idea of how Betaflight (or open source in general) development works, with most if not all developers contributing code also holding down day time jobs...

adamtfc87, 2018-01-23:
@mikeller I apologize, I guess I was just looking for some kind of confirmation that it might be considered for the future. Totally not expecting it to be rolled out in a week, or even months! Thanks for all you guys do, I don't understand a whole lot about open source development or software development in general, but I do know it's a lot of tedious and hard work and I appreciate what you guys do :)

mikeller, 2018-01-23:
All good. When it comes to feature requests, 'no news is good news' applies to a large extent - if nobody pipes up with reasons why this is not doable, or why it should not be done, you can consider your feature request as silently accepted, waiting for somebody to do the work.

closed 2018-11-10 by mikeller