betaflight/betaflight #4968: [Feature Request] GPS speed displayed in MPH

adamtfc87, 2018-01-15:
Wondering if gps speed on the OSD is possible to be displayed in miles per hour. Thanks!

DanNixon, 2018-01-15:
Yes, this is fairly easy.

It is likely going to also require the addition of the "UK" unit system (i.e. Celsius and miles).

mikeller, 2018-01-15:
> It is likely going to also require the addition of the "UK" unit system (i.e. Celsius and miles).

I think now we are going over board. Getting speed in a unit that one is familiar with is valuable. Core temperature is purely informational...

DanNixon, 2018-01-15:
> I think now we are going over board.

That was just an idea based on what iNav do (seemingly with less reason as all their units are distances).

mikeller, 2018-01-15:
They do have US / UK / EU?

DanNixon, 2018-01-16:
Yes (they call it imperial/UK/metric): https://github.com/iNavFlight/inav/blob/master/src/main/io/osd.h#L68-L72

I'm not too sure of their definition of UK units though, I'd always thought anything derived from distance was imperial (i.e. distance and speed), they only show speed as imperial when UK is selected.

mikeller, 2018-01-16:
The plot thickens...

Maybe @martinbudden can enlighten us here?

jflyper, 2018-01-18:
Aviation use nautical miles, right???

mikeller, 2018-01-18:
@jflyper: Yeah, I thought so too - and feet for altitude.

martinbudden, 2018-01-21:
I seem to remember the iNav discussion about this. A few people wanted height in meters, but distance and speed in miles.

This is basically what we have in the UK. Everything is metric with the exception of road signs which still have distances in miles and speeds in mph. The trouble with converting road signs is that it is very expensive, and the transition period is problematic.

Many years ago I went cycling in Ireland when they were in the process of converting. Some road sign distances were in miles and some in km which was confusing/amusing. I believe they have fully transitioned now.

lazd, 2018-03-07:
I consider it a bug that, when `osd_units = IMPERIAL` is set, the OSD display reads speed in kilometers per hour (a metric unit), but other values are displayed in imperial units.

Barring implementing a imperial/UK/metric option, the existing imperial option should display speed in imperial units.

[image]

zytra, 2018-03-12:
I have the same issue as Lazd - changing from metric to imperial doesn't change the value displayed on the OSD (K = km/h) - verified the value/unit in my car at constant speed.

it would also help changing the unit on the OSD to either show "km/h" or "mph", as "K" kinda of suggests "knots" or "kts".

cheers..

lazd, 2018-03-12:
@zytra AFAIK the `km/h` and `mph` symbols don't exist in the font any longer, so they will have to be added there first.

zytra, 2018-03-13:
I guess what I'm trying to say is that we don't need an additional image in the font, but simply adding either the 3 letters "MPH" or the 3 letters "KPH". I know it will take a little more space but I would think that's no big deal. Either way, I the unit option "metric"/"imperial" does not change anything to the value displayed as of 2 days ago.

Cheers

neilbalch, 2018-05-04:
Is there a reason why the units cannot be set so mixing and matching different sets is possible? Seems like it would be easier for everyone if we didn't have to create a new preset per combination someone desires.

neilbalch, 2018-05-04:
@mikeller  Yeah, I do believe that 64KB might qualify as a small flash size. ([digikey.com/catalog/en/partgroup/stm32-f3-series/31232](https://www.digikey.com/catalog/en/partgroup/stm32-f3-series/31232))

etracer65, 2018-05-05:
Note that #5765 fixed the original issue - that GPS speed wasn't respecting the units setting.  Now it will be displayed in MPH if imperial units are selected.

IcyLemur276, 2018-05-13:
@etracer65 what firmware version is the fix in? 3.4?  I don’t see the fix in 3.3.2.

etracer65, 2018-05-13:
@IcyLemur276 It's include in 3.3.2 as noted in the release notes:

> - Changed GPS speed and max speed to respect the units selection (#5765);

IcyLemur276, 2018-05-13:
@etracer65 I can confirm the fix doesn’t work for 3.3.2.  Still shows “k” for speed while other units are imperial.

etracer65, 2018-05-13:
I just tested on 3.3.2 and it properly displays the units in MPH (with a "M" suffix) in the OSD.

IcyLemur276, 2018-05-13:
@etracer65 I’ll investigate again later. Maybe it’s my fonts as I’m not using defaults but custom fonts.  Hmmm

IcyLemur276, 2018-05-13:
@etracer65 you are indeed correct.  Display now shows "M" instead of "K", but in the OSD Betaflight Tab it shows a K there still, but as long as i'm seeing M in my goggles I'm ok with this.

closed 2026-09-26 by blckmn