betaflight/betaflight #4977: Betaflight 3.3 Kalman extended roll right

KnuckleUpFPV, 2018-01-17:
Betaflight 3.3 Kalman Filter activated. Followed the post Joshua Bardwell linked. 

Board is matekf722
32/8 loop speeds. 
This firmware fixed its previous yaw twitch

New problem: 

If you roll right more than once, the flight controller will complete two full rolls and stop itself completely level. These additional rolls are uncommanded. Throttle and yaw input still work during this event. Pitch and roll have no response. 

Roll left does not experience this issue. It happens every time I try it. 

Super Rate is set to .84 for roll,pitch, and yaw. Other than that, it is superb.

mikeller, 2018-01-17:
What build of the firmware did this happen with?

KnuckleUpFPV, 2018-01-17:
This is from the drop box link on rc groups. I believe bardwell said its rs2k's build. I just went to the post bardwell linked and followed the guide.

Jahnkeanater, 2018-01-17:
This is an issue with RS2K's BF 3.3 fork.
https://github.com/rs2k/betaflight
https://www.dropbox.com/sh/ijtkl9bfu4pgq8z/AAACQugd9iSqkkPLDT1owgY-a?dl=0

The new filter is not part of the betaflight code. I think this fork is based on a build from January 2nd.

mikeller, 2018-01-17:
@KnuckleUpFPV: From now on, please add a the build number, or a link to the source changeset that is used for the build, to feedback about this build. Trying to discuss a particular build without having access to its sources is pretty much pointless.

closed 2018-01-17 by mikeller

Jahnkeanater, 2018-01-17:
Betaflight 3.3.0 Jan 12 2018 / 11:19:35 (41cc680) MSP API: 1.37
Could the flip issue be caused by the gyro overflow code?

mikeller, 2018-01-17:
The commit that this was built from is 28 days old. Have you tried updating to a more recent version.

KnuckleUpFPV, 2018-01-17:
@Mikeller. Had I known the build number I would have posted it. All I know is it came from rs2k's drop box. I posted all the info I had. This is the firmware bardwell posted with directions on how to use it. If for nothing else it may help the other thousands of people that watched his video and are going to use this firmware. On the betaflight facebook page they asked people to try this firmware with the kalman and report issues. And try the newer firmware with the biquadrc fir2 and compare the two. And report any issues. Anything else I can do for you?

Jahnkeanater, 2018-01-17:
@KnuckleUpFPV download and test biquadrc fir2 from this link and see if it fixes the issue.
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1964/artifact/obj/betaflight_3.3.0-1964-bca7b90.zip

KnuckleUpFPV, 2018-01-17:
Will do. I'll test tomorrow

mikeller, 2018-01-17:
@KnuckleUpFPV: If you want to find the changeset a firmware was built from, doing a `version` in CLI will give you a line like

> Betaflight 3.3.0 Jan 12 2018 / 11:19:35 (41cc680) MSP API: 1.37

Posting this is one way to identify the changeset, as the value in braces is a unique changeset id.