betaflight/betaflight #4981: WHY ????

cephi58, 2018-01-18:
Please stop with that high frequency update policy for the configurator. You're gonna ruin the hobby !! There are peole whose time is spare and a configurator should be a tool and not a challenge!

It cannot be that each simple and in this case mandatory update costs hours of research and problem solvings (window10 issues) e.g. only for re-positioning an OSD feature in an existing quad. And changing the way of installation for every release also costs learning time (means for W10 sometimes significant effort as the defender does not like it all too much anyway).

Defaulting to OS language without interrogating the user is not good either. Especially not if it is only an update of the existing configuration. Most of the people probably DO want to have it in English anyway simply because it makes internet research much easier. The information that the language could be changed but not mentioning how and where ist simply just annoying and in no way helpful at all. Still haven't found out how to do it so far!

Additionally to be mentioned is that the OSD feature does not work correctly anymore for 16:9 (RC Eagle V2) cams. You cannot position anything below rougly the middle line of the screen. This was not the case in former releases. And the user's former lower OSD value configuration is left somewhere in the smithereens way below the screen and therefore can neither be grabbed nor moved. You have to wipe out everything and re-configure. You have to position everything in the upper half of your FPV image this way what I personally do not like.

Cheers
Tommy

mikeller, 2018-01-18:
Hi @cephi58.

A couple of points:
- you posted this issue to the issue tracker for the Betaflight firmware. This is a configurator issue, so next time please use https://github.com/betaflight-configurator/betaflight/issues;
- when you are opening an issue, please use a title that is indicative of the content of your issue - other users don't like to have to read through issues just to find out if it's something that is of concern to them or not;
- when it comes to the recent configurator updates, you have to understand one thing: We are not doing this by choice. The move away from the google chrome web store that we've used for years is caused by the fact that google will end support for it, and for google chrome apps on desktop operating systems, later on this year. We embarked on the journey to move to the standalone apps last October, and an a considerable amount of time and effort had to be sunk into bringing the standalone apps to where we are now - and we still don't quite have the quality and ease of use that we would like. That is not even taking into account the application bugs that had to be fixed and the updates to support the latest features in the firmware that had to be added.
One goal that we have - and one area that we still have to improve - is to make updating the configurator similarly seamless as it was back in the chrome web store days. Bear with us, this is not easy, particularly not if you are new to it;
- defaulting to the OS language is what the vast majority of applications do - and this includes the framework that we are using to build the standalone apps. We understand that users wish to keep the app in English, and we are committed to supporting this - in the last release it did not work out to be as easy (i.e. a separate app icon to click for English). Again, we are learning as we go;
- can you please open an issue in the betaflight-configurator for the OSD problem that you are reporting - we'll get this fixed in the next release.

cephi58, 2018-01-18:
@mikeller 
Mea culpa! Sorry for the misplacing and the useless title. I fully agree on that point. And yes, the Google policy regardiing DTs is a pain in the xxx  these days. Not only for you guys ..... Reason probably is that MS will go 'Android' for free on non-business (Home Version) Windows users.

The link provided by you does not work (404 Error). Feel free to wipe my post out of your thread and - if you to do so - place the relevant things from above to the place where they belong. I'm pretty sure that the language and the OSD issues are also relevant for some other users and might deserve some attention.

Btw, the goal of the posting was not to blame anyone but give feedback as a user (or better here: tester :-) ). Honestly for the last 3 builds it took me more than double the time for the BF configuration than for the build themselves. But 32-bit driver issues on the latest 64-bit W10 with USB3 ports are generally a nightmare

mikeller, 2018-01-19:
Typing URLs by hand when tired isn't a recipe for success: https://github.com/betaflight/betaflight-configurator/issues.

Thanks for the feedback. And yes, you _are_ being used as a tester. After all, this is a hobbyist project, and we only got the money that our users pay us to pay for professional testers. :-P

We are aware that the driver situation is exacerbating the difficulties for many (mostly Windows) users. We are going to look into integrating driver support into the installer. But the same thing applies there: Expecting us to manage to make it work better than others who have been doing this for a while (Zadig,...) manage is probably not realistic, so it might get worse before it gets better...

Bluntrollindboy, 2018-01-19:
why though?
you made me feel like i was insane
@mikeller

mikeller, 2018-01-19:
@Bluntrollindboy: Why what?

cephi58, 2018-01-20:
@mikeller 

Just my five cents to all those topics that obviously occur these times around FC configuration and provider

"Configurator"
If your configurator does not work you will not fly (at least not on BF). Full stop.

"Hobbyware"
This is actually not the whole truth anymore. Many people get money from 'Betaflight' chain . It is fastly moving towards a comercially established brand. For my part, it goes from backpacks to ESC or FC hardware. This means that - if the configurator does not work correctly or is acessible - a significant amount of invested money is at stake and the former decision to go for Betaflight could easily turn out to be fundamentally wrong. With market, influence and,  as a consequence, money also comes resposability and duty. Nearly all the influencers on YouTube (like Bardwell, UAV Steward, or the late Blue Falcon) USE Windows and explain (and therefor spread) your stuff are doing that based on Windows. And they are full professionals nowadays who make their living out of it. We are also your clients.

"Windows Guys"
Yeah we exist and partially not voluntarily. Having an own IT company and clients in the financial branche does not give you full freedom of choice of your OS (and it's defaoult language). MS is - unfortunately - the widest spread operating system in the world. If you want to grow and establish you have to take that into account. W10 is crappy malware. You (and I) have to live with it. No way not to do so. If you force users to have a sepratae system for the hobby you will loose many of them.

"Language"
This is clearly a nice to have and not important. In 35 years working for the branche as a techie and consultant I have never seen any technical application with english termini that has become successful in translated versions. The termini are nearly always the english ones and you do research in english based on this. Translating twive significantly reduces information precision and means additional effort or sometimes even impossibility. And you do no care about language if you cannt connect anyway. Adopting tool tip using os language with english expressions would do the job

"Client site Testing"
Yeah, I'll do so, if you want me to do. Voluntarily. Just tell me what I can do for you. You have my mail address. I'll give you my real name and address, my configurations and my HW possibilities. Provide me with the cases. I'd like to contribute because I liked Betaflight so far. And sure it does not go without us users for the time being. But do not force me to updates that lead to the fact that I cannot fly anymore or costs me hour or even days without announcing it. It's not only be me who will be annoyed. I am just the first to state it clearly here (and take the blame :-) 

Hope you take that as supportive critiscism and not blame.

Kind regards
cephi58

mikeller, 2018-01-20:
@cephi58: I don't quite get what you want to say here. Most of your statements don't seem to be related what was said above. Is it just that there was nothing that you wanted in the latest configurator release, and that you'd preferred for this not to be released? That's your personal opinion, and you're entitled to it.

(Also, your statement about not being able to fly when the configurator does not work is simply wrong. There is also the CLI: https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md)

cephi58, 2018-01-20:
@mikeller 
As we both figured out already this thread is in a wrong place but I cannot move it from my side. I already begged for pardon for doing so. I cannot tell from my side whether it is a configurator or a firmware issue (or even a hardware incompatibility to W10 64 latest issue board side). I just can tell that it occurred somewhere in between 10.0.0 and the next one because it is the only thing that obviously changed. (btw, same problem with BF3, LUX4, BF4 FC). My personal 'feeling' is that MS eventually does not support 32 bit drivers anymore at all in the future because they cause troubles for them somehow (or cost and effort).

And no, it has nothing to do with desired additional features. It is that it does not work after the last bigger updates of W10 64 or BF configurator. Neither with or without Zadig. Not only at my site! You cannot connect to your FCs anymore. There is a work around  if you manually de-install ALL the W10 drivers, start it up again without drivers and manually re-install all drivers without Zadig (or Impulse RC driver fixer or whatever tool) which means time AND risk !! Be sure that you do not uninstall the wrong parts!! But as soon MS makes a new release you would probably have to redo it as MS kicks it out again. Might be a security issue or OS problem ...  You have to ask MS. And yes CLI probably would work eventually somehow if you COULD connect but it would then be a clear step back and means significant know how getting effort.

For me BF as FC and config ware was perfectly all right as long as it was working. I did blind tests with new 3.2 stuffs with all my collegues I am building and configuring quads for and none of them was able to figure out ANY difference with the new features compared to 3.1.7 and neither did I. At least not so much that i would risk system stability for it. Frankly none of them noticed even any difference between DSHOT600 8/2 and DSHOT1200 8/8 with or without dynamic filtering. One person even said that it was better only because I have told him I'changed it without doing anything. This is called placebo effect. There might be one for pros but not for us weekend hobby flyers

I ONLY wanted to bring into consideration why (and that) stabilizing the software and adapting it to NON LINUX (or Apple) - or in my case explicitely W10 64bit latest release environments should be priority number 1 as Google is not available any more and the stand alone does not work for us since the update away from the first release (10.0.0 as far as I can  remember). For whatever reason this might be.

Once again it was meant as helpful hint and not blame anybody for anything

mikeller, 2018-01-20:
Ok. To me it looks like your main gripe is with the Windows drivers that are used by the configurator. As I already stated earlier, we are aware of the problematic situation there, and looking into ways to improve it.

closed 2018-01-20 by mikeller