betaflight/betaflight #4984: OSD Warnings: Break out individual warnings and add "failsafe" warning

EggsBenedict, 2018-01-18:
The new OSD features are amazing and so is the "Switch All" toggle.

I do wish the catch all for the "Warning" slider could be modified so I could **keep some warnings** and not others **maybe these could be broken out into a sub category each with their own slider**?

**And a warning I'd like to see added if possible is one that indicates "FAILSAFE STAGE #"**

I know it seems silly ... but sometimes with super quick failsafes ... i think people brush it off as a maybe an FC hiccup or just other issue that doesn't raise the red flags of "there could be serious issue or interference that's going to lead a stage 2 failsafe soon, maybe I should land."

DanNixon, 2018-01-18:
> I do wish the catch all for the "Warning" slider could be modified so I could keep some warnings and not others

You can already do that. See the "Warnings" section of the OSD tab.

EggsBenedict, 2018-01-18:
oh... maybe i missed this because it doesn't seem to work on my version of betaflight screen as shown below. Am i doing something wrong? Toggled everything off and on and then just Warnings alone but no joy.

[image]

EggsBenedict, 2018-01-18:
upgraded to 3.2.4 just to confirm that wasn't the issue. same outcome as above.

DanNixon, 2018-01-18:
The ability to toggle warnings was added in 3.3 (see #3924).

EggsBenedict, 2018-01-18:
Ah ha! ok! that's awesome! Sorry - i did search around - but failed obviously.
should i close this issue and add a comment to that one re: failsafe warnings? or leave this as is?

DanNixon, 2018-01-18:
No problem.

You can keep this one open and change the title and message to reflect the request for a failsafe state warning.