betaflight/betaflight #4988: Matek F405-CTR led strip not working

ellingo, 2018-01-19:
I have two Matek F405-CTR boards and the LED-strips will not work. 

With the LED_STRIP on, they initializes with a white flash. After that I on one one board get the colors, but no blinking. On the other board I just get something not resembling what I have set. 

With the LED_STRIP off, the white initiation flash is missing, but otherwise no change.

I also have one Matek F405-AIO which is working with the same firmware file.

mikeller, 2018-01-19:
@ellingo: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20led%20strip%20diode).

closed 2018-01-19 by mikeller

richard-scott, 2018-01-19:
That's a strange way to design a group of F4 FC's then... Changing the pin outs on each board variant but have it based on the same BF Firmware which has it fixed on a set pin. At least the Matek website documents it if you an get to it.

ellingo, 2018-01-19:
Thanks, found it. One board is running perfect now.

mikeller, 2018-01-19:
@richard-scott: The way Matek designs their boards, re-using the same pinout for the active components on boards with different form factors / features is actually quite good - it removes the confusion that normally arises when there are multiple firmware variants for similar boards from the same manufacturer.

ellingo, 2018-01-20:
All good, I hope this thread will help other people.  
Go here: http://www.mateksys.com/?portfolio=f405-ctr#tab-id-3
Everything is found under the "BF FW & Wiring" and "BF Pins Definition" tabs.

Phonicswave, 2018-02-27:
Hay, i led`s dont work on S5 ( Resource A15). On the matek website it says "if you want to use the barometer and the led strip at the same time, you have to use the S5 pin (resource A15) for the led strip"

„2812LED_Strip share I2C1_SCL pad as default.  either enable LED_Strip, or enable Baro.
2812LED can be resourced to S5 (PA15) if using Baro the same time with BF firmware“.

So i wire up the led signal wire to S5 pin and write the following cli commands.

- Resource Motor 5 none (because motor 5 is set to S5 pin = resource A15)
- Resource led_strip 1 A15 (betaflight confirmed it to me)
- Save (reboot)

Connected the battery and.. nothing. led strip doesn`t work.

So, to see if the the led works i wire up the signal wire from S5 back to the led pin on the fc and set it up in betaflight (Cli command: Resource Led_strip 1 B06) Betafligt say`s B06 is already in use „Note: B06 already assigned to I2C_SCL 1“ (Baro) but confirmed it.

The Baro is still enabled in the configuration tab in Betaflight.

Connect the battery, now it works.. not really.. but the led´ s shine in the color i configured it in betaflight.
As long as i enabled the barometer (configurator tab) i can not switch off the and on the led`s (betaflight - modes - ledlow) so i think the barometer doesn’t work too.

I also tried different firmwares.

My question ist, Can i rout the barometer anywhere else? Or does anyone have another solution to use the baro and the led`s at the same time? S06 is already in use for camera control. Is this a software issue or is my board broken, or Maybe my routing is wrong?

Thanks

ellingo, 2018-02-28:
Moving the LED-strip to S5 works on some boards, but not on others. I believe there must be two revisions of the F405-CTR, but they are hiding this fact from us. Cannot know for sure, though, so it is too early to go postal...
On boards where it doesn't work one must turn off either the barometer or the LED-strip in Betaflight Configurator, if you are not able to move the LED-strip somewhere else than S5.

ellingo, 2018-02-28:
Oh, well. Received a board today and it doesn't work on this one. I have now tested four F405-CTR boards in total and LED-strip on S5 works on only one of them.  The next hypothesis is that there is a glitch in the Betaflight code.

jflyper, 2018-02-28:
> The next hypothesis is that there is a glitch in the Betaflight code.

Can't happen.

jflyper, 2018-02-28:
I have tested MATEKF405 target with a bare dev board and PA15 assigned as LED strip output works without any issue.

> I have now tested four F405-CTR boards in total and LED-strip on S5 works on only one of them. 

Logical consequence is that there are F405-CTRs that LED strip on PA15 works and those that don't.

jflyper, 2018-02-28:
One more test to do is to power the LED strip from 3V3, if you are powering it from 5V.

ellingo, 2018-02-28:
> Can't happen.

Famous last words from all programmers I have known. Or: "It works on my computer."
Anyhow, I will not report this as a software bug. If there is one it is faster to fix the bug myself and build it.

ellingo, 2018-02-28:
> One more test to do is to power the LED strip from 3V3, if you are powering it from 5V.

Done that

jflyper, 2018-02-28:
Then I would start poking around the board with my DMM, starting with checking connectivity of PA15 to S5 pad.

ellingo, 2018-02-28:
> Then I would start poking around the board with my DMM, starting with checking connectivity of PA15 to S5 pad.

Yeah, but I find it hard to believe I have exactly the same hardware problem on 3 of 4 boards. However, nothing can be excluded as one cannot know what the problem is before it is fixed.

MATEKSYS, 2018-03-01:
check the tips here.
http://www.mateksys.com/?portfolio=f405-ctr#tab-id-4

jflyper, 2018-03-01:
I would suggest clearing configuration by CLI "defaults", and when the board comes up again, immediately configure the LED strip on S5.

Phonicswave, 2018-03-01:
I did it all again. first "defaults" after reflashing (3.3_RC3) and did the other cli commands.

Led's still dont work.

Then i measured the voltage between the pins and i think i found the problem.

(Ground to 5V pin = 5.0V)
Ground to led pin = 3.3V (signal wire for the led`s)
Ground to S5 pin = 0.0V (I think that should be also 3.3v because this is, if you set it up in betaflight, the new signalwire pin for the led)

I tried to bridge the S5 pin with the 3.3V pin, but that does not work either.

Can anyone, who owns this board where the led`s work on s5, please measure the voltage between ground to the led pin and then ground to s5 pin? With a working board I think there should be 3.3V on the S5 pin. 

Can someone confirm that?

thanks

jflyper, 2018-03-02:
> Then i measured the voltage between the pins and i think i found the problem.
>
> (Ground to 5V pin = 5.0V)
> Ground to led pin = 3.3V (signal wire for the led`s)
> Ground to S5 pin = 0.0V (I think that [...]

These measurements actually mean very little or none, especially without an information about a configuration you are running.

Can you paste outputs from CLI command diff, right after you did `defaults` and configured only LEDs,  please?

jflyper, 2018-03-02:
Hmm... How about `resource list` and `dma`?

Phonicswave, 2018-03-02:
**resource list**
Currently active IO resource assignments:
(reboot to update)

A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FLASH_CS
C01: SDCARD_CS
C02: MPU_CS
C03: MPU_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D02: FREE

Use: 'resource' to see how to change resources.

**dma**
Currently active DMA:

DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Thank you for trying to help me :)

AndersHoglund, 2018-03-02:
I browsed thru this issue and I can not find that anyone has mentioned the well known fact that a LED-strip usually requires a diode in series with the 5V supply, to lower supply voltage and minimize the diff between high signal level and supply voltage. There was some suggestions on running the LED from 3.3v that would point in this direction, but the 3,3V regulators are usually to weak to handle supply current demanded by a LED -strip. 
RTFM: https://github.com/betaflight/betaflight/blob/master/docs/LedStrip.md 

> If you have LEDs that are intermittent, flicker or show the wrong colors then drop the VIN to less than 4.7v, e.g. by using an inline diode on the VIN to the LED strip. The problem occurs because of the difference in voltage between the data signal and the power signal. The WS2811 LED's require the data signal (Din) to be between 0.3 * Vin (Max) and 0.7 * VIN (Min) to register valid logic low/high signals. The LED pin on the CPU will always be between 0v to ~3.3v, so the Vin should be 4.7v (3.3v / 0.7 = 4.71v). Some LEDs are more tolerant of this than others.

jflyper, 2018-03-02:
@Phonicswave 
```
resource list
Currently active IO resource assignments:
(reboot to update)
...
A15: FREE
...
```

Your board is not configured to work with LED strip on PA15; neither of pin or DMA is assigned, even it is listed in the `resource` output.

## I suspect you haven't turned on `LED_STRIP` feature.

Phonicswave, 2018-03-02:
It works, thank you soo much, all of you.

I power up the led strip via the 3.3V pin now.
I`am feeling soo dump right now, yesterday i tried to bridge the 3.3v pin with the led signal wire S5 pin..for no reason.

**jflyper** already said 2 days ago try to power up the led`s via 3.3V if you are using 5V, this was already the solution.

Whatever, now i can use the Led`s and the barometer at the same time, via resource A15, S5 pin.

Thanks!

ellingo, 2018-03-02:
Yes, thank you all. I am now using a 1N4007 from 5V to the LED strip and it works on at least of those three boards.

Can anyone tell me why the LED strip works when connected to the LED pad without a diode, but when connected to S5 it needs a diode? I assume there must be a difference in the output circuitry of PB6 and PA15, but why? I understand we all tend to use our LED strips in an out of spec way, but I am curious.

ellingo, 2018-03-02:
Note that to drive one WS2812B requires up to 150mA. The 3.3V output can strictly speaking only drive two LEDs safely. All three LEDs (reg, green and blue) full on.

ellingo, 2018-03-02:
According to the datasheet for WS2812B the minimum voltage is 3.5V.
Because of the current draw and minimum voltage, a diode between 5V and the strip seems to be the best option.

MATEKSYS, 2018-03-04:
@ellingo  Try to power it via 4V5 pad,   5V-diode-4V5

jflyper, 2018-03-04:
> Note that to drive one WS2812B requires up to 150mA. The 3.3V output can strictly speaking only drive two LEDs safely. All three LEDs (reg, green and blue) full on.

I think you are confusing DIN drive capacity with power consumption of a single WS2812B device. LEDs inside the device is powered by VDD, independent from DIN.
DIN drive current (II : Input Current) is rated at only 1[uA].

Further notice that MCU signal pin only has to drive DIN of the first device. Other devices will be driven by DOUT of previous device in a row.

BTW, as far as I understand, each device's LED consumption is somewhere around 20[mA/color] or 60[mA/device] at the most. I wonder where you got 150[mA] figure from.

> According to the datasheet for WS2812B the minimum voltage is 3.5V.

Again, I think you are confusing supply voltage and signal voltage.

Note that there are several different ratings and conditions.

(A) VDD (Supply Voltage) 3.5~5.3[V]is the rating for Vdd (Supply Voltage).
(B) VI (Input Voltage) -0.5~VDD+0.5[V]
(C) VI also have to satisfy 0.3VDD >= VI for logic low and VI >= 0.7VDD for logic high when signaling (VIH and VIL conditions)

When powered by VDD=5V and signaled by 3.3Vpp, these would satisfy (A) and (B).
However, (C) is marginal, because 0.7 * 5.0 = 3.5[V] which is higher than signal peak voltage.
For signal voltage 3.3[V] to satisfy (C), VDD should VDD <= 4.7[V].
Fortunately, it is working, in most cases, thanks to slightly lower VDD and/or wide margin of the device.

If both of these luckiness is not in effect, the device wouldn't work.
This is why it is advised to supply VDD from 4.5[V] above.

Hope this helps.

ellingo, 2018-03-04:
@jflyper No I believe I am not confused. I placed a 1N4007 to lower the power supply voltage to 5V-Vf where Vf can be estimated to be 0.6V but varies according to device, current draw, temperature, etc. So, using this rectifier diode to drop the voltage I can draw 1A to power up to 6 x WS2812B.

To power the LED strip from the 3V3 pad is not guaranteed to work as the voltage is lower than the 3.5V required. It can also power only 2 x WS2812B.

To power the LED strip from the 4V5 pad is possible, but only if it can supply the needed current. Which seem to be unspecified.

Therefore it seems best to use the 5V pad to power the LED strip, but with a rectifier diode inserted in the power wire to create a voltage drop of 0.6V.

ellingo, 2018-03-04:
@MATEKSYS What is the current rating for the 4V5 pad?

ellingo, 2018-03-04:
@jflyper I might add that a LED strip working when controlled by the LED pad tends to not work when controlled by the S5 pad.

ellingo, 2018-03-04:
@jflyper About the current drawn by one WS2812B I read:
"Each RGB LED draws approximately 50 mA when it is set to full brightness and powered at 5 V."
I interpreted this meaning 50mA for each of the three LEDs in one device. I can see my interpretation may be wrong. In that case all my current draw numbers should be divided by 3 which opens for some possibilities.

MATEKSYS, 2018-03-05:
@ellingo  5V -- B5817W 1A Schottky--4V5

ellingo, 2018-03-05:
@MATEKSYS Thanks. If one WS2812B draws 60mA, as mentioned by @jflyper, then this should be enough for a looong LED strip. However, I think I will use a separate diode between 5V and the LED strip. That way the lights will be off when the board is powered via USB.

Thank you for being here in the forum, MatekSys.

guelz, 2018-03-13:
Works with a diode! Thanks

organism-46b, 2018-03-22:
same issue with F405-STD not working on S7
i have 1N4148  would it work?

richard-scott, 2018-03-23:
@organism-46b on the F405-STD the documentation says the LED Pin is S5, not S7.
http://www.mateksys.com/?portfolio=f405-std#tab-id-4

organism-46b, 2018-03-23:
http://www.mateksys.com/?portfolio=f405-std
 under "tips" line 5
[image]

alexz-kh, 2019-05-07:
Had the same issue at fw v3, and fixed it with recommended here :
```
batch start
Resource Motor 5 none
Resource Motor 6 none
Resource Motor 7 none
Resource Motor 8 none
Resource led_strip 1 A15
batch end
Save
```
But, after upgrade(with full erase, for sure) to version:
```
# Betaflight / MATEKF405 (MKF4) 4.0.2 May  5 2019 / 12:02:40 (56bdc8d26) MSP API: 1.41
```
Work-around not work anymore. any idea?