betaflight/betaflight #4990: Full throttle yaw with Biquad RC+FIR2 Filter

Jahnkeanater, 2018-01-19:
Betaflight / OMNIBUSF4 (OBF4) 3.3.0 Jan 18 2018 / 12:44:40 (c0b8f6c7d) MSP API: 1.37

As soon as you increase throttle slightly the quad violently yaws and goes full throttle.

I have tried setting the q and r values

set gyro_filter_q = 0
set gyro_filter_r = 0
and
set gyro_filter_q = 300
set gyro_filter_r = 80

Does setting the values to zero disable the FIR2 filter?

fujin, 2018-01-19:
Yes, when both values are zero it is disabled.

TheIsotopes, 2018-01-19:
@Jahnkeanater seems to be the same problem as mine #4989

Jahnkeanater, 2018-01-19:
Just tried reversing yaw direction and it seemed to hover for a second before going into a full-throttle spin. Its like the yaw direction randomly changes. I think there is a bug in the new rates code possibly.

Jahnkeanater, 2018-01-19:
Hovers with inverted yaw 

set yaw_control_reversed = ON

yaw drifts in a clockwise direction as soon as the quad yaws left it violently yaws counter clockwise.

RipperDrone, 2018-01-19:
maybe set yaw_control_direction is set wrong?

mikeller, 2018-01-19:
Ok. One thing that comes to mind: Did you guys do a 'reset to defaults' after flashing the firmware with RaceFlight rates, and then a restore of the config?

Not doing the reset will mess up the config (including throttle rate / expo), because the structure / number of parameters has changed.

closed 2018-01-20 by mikeller

Jahnkeanater, 2018-01-20:
@mikeller Are you talking about the restore button on the first tab?

From defaults, I pasted my settings into the CLI. Then I go back through the tabs just to make sure the settings are correct. I could test default settings just to make sure.

richard-scott, 2018-01-20:
@Jahnkeanater After flashing and before pasting your config back into the CLI type 'defaults nosave' and then paste in your config and save.

PR #4992 (2018-01-20): Fixed Betaflight rates calculation. Moved `rates_type` into rate profile config in CLI.
Fixes #4990.