betaflight/betaflight #4991: VTX setup used from Spektrum Tx causes FC to hang intermitantly

AndersHoglund, 2018-01-19:
This issue is just to make a note on using VTX setup from a Spektrum Tx is unstable. 
Problem seen with DYSF4PRO, Tramp HV, SPM4649T and Dx18(g1) .
Typical symptom is that FC hangs/crashes with beeper going off. No way to recover but to reboot.

I am in the process of trying to bisect testing all of this, trying to find where this started to occur.  Early commits are OK ( if 50 or so test runs are OK) while later commits fail with one (1) test. Only changing channel causes FC to crash. No real conclusions as yet what might be the cause, this is just an early warning ( Well. not early really. It has simply not been tested for quite some time.) To complicate the matter, I have run 40 or more test without a problem on early revs, only to find test no 51 fails... But the fact is, later revs fail much faster.   To be continued....

AndersHoglund, 2018-01-20:
Bisect testing proved to be inconclusive. I can get some late revs to fail on first try, and some early not to fail at all. But then some revs in between fail at 25 tests and some at 51 etc in a random pattern.
And now on the current git HEAD ( commit db0f1c4f10b9403847904a623d919957885af7be) I can not get any failures at all (or ~55 tests OK).  But still, no real explanation what caused the hangups. 

I'll keep this issue open for observation for some more time. A bit too early to close it and say "Fixed". 
Would be nice if someone else with similar setup could chip in with some test results.

AndersHoglund, 2018-01-20:
This is a real problem.
When using a BETAFLIGHTF3 FC I can get it to fail much quicker. One to four tests to change a VTX channel will eventually hang the FC with beeper going off on current HEAD rev. 
Reverting to initial rev (37608a0) with first "Spektrum VTX Setup" support is OK, 25 tests and no hangups. Bisect will commence.

mikeller, 2018-01-20:
@AndersHoglund: Not really related to your bug hunt, but as a heads up: The changes in #4982 might be useful to get better separation of RX update checking and data processing for your VTX code.

Also, good luck!

AndersHoglund, 2018-01-20:
Thnx, that is interesting. Will take a look at it.
Bisecting gave some interesting and confusing results.....to say the least. The last rev to function properly is (625f083a3fb7cbc4bf26b0492cfaef265aac3099) . Next rev is PR #4722 and then the merge of that PR, both failing.  But this is puzzling, how can that PR affect VTX setup? Have not looked deeply into what that PR accually does. But I did rebuild and retest the commit immediately prior to that one and there was absolutely no problem when running that on BETAFLIGHTF3, Twice. First time 25 times OK. Then the PR 4722 commit and the merge, failed after 8 and 3 test runs. Rebuilt and retested the passing rev again, 50 times OK.
In a timeline it looks like this:

```
commit 625f083a3fb7cbc4bf26b0492cfaef265aac3099 ----------> PASS built 25 channel changes OK
commit f9feb03a93d73b8aec325e9b8b668062dbe11bf7 Merge of PR 4722 --> Fail after 3 VTX channel changes.
commit b9272ae325f58df123a7a04840822e58455ca53e PR 4722 itself ----> Fail after 8 VTX channel changes.
commit 625f083a3fb7cbc4bf26b0492cfaef265aac3099 again -----> PASS 50 VTX channel changes.
```
Do a `git log f9feb03a93d73b8aec325e9b8b668062dbe11bf7` to see the full history and change notes.
And it is confusing, I can not see the connection between 4722 and VTX control. I did of course suspect my own VTX and Spektrum hacks, but no. Will dig some more to see what sideeffects either of these two separate items might have that may cause this interference. Stack usage/overflow, memory leaks, wild pointers etc etc etc,  I could use some help and ideas ..... 
Oh, also tested PR #5000 in hope it would resolve the issues, but sorry to say it did not. Still failing.

AndersHoglund, 2018-01-22:
I am not really getting anywhere with this. Well, I did one experiment that showed some promising results. 
As of now the Spektrum VTX Setup or control code is running in the RX tasks. I made a try to move all or most of it into the VTX_CONTROL task. There are some logic in that.... But having vendor specific VTX input control code and generic VTX output control code together looked a bit odd. But still, VTX input to VTX out control code is communicating via EEprom config settings. Writing and reading EE-prom might be too slow (have not measured yet) to be run in the RX tasks. Will do some more experiments, this is just a note to raise the Q and the thoufgts around this. Input please.
Next step will be GDB sessions, one BFF3 prepared for SWD....

AndersHoglund, 2018-01-25:
A few notes to summarize what has been done and what has been seen.
GDB sessions gave nothing. No problems seen with non-optimized, feature striped and debug-able code. Normally built code can crash even in GDB, but not analysable, no symbol info. 

When crashing the typical sideeffect or symptom is that the beeper goes off continuously. Even when not crashing I have heard the confirmation beep for VTX ch change being cut of. Did try to use both 1, 2 and 5 confirmation beeps and I have heard 0.5 and 1.5 beeps. Something is cutting of the `multibeeps `used.
There is a note on multibeeps: 
`src/main/io/beeper.c:    { BEEPER_ENTRY(BEEPER_MULTI_BEEPS,           13, beep_multiBeeps,      "MULTI_BEEPS") }, // FIXME having this listed makes no sense since the beep array will not be initialised.
` 
Not sure this note makes any sense to me, nor what todo about it or how to avoid it. 
Beeper code is a suspect... Re-entrance problems? Can not see anything obvious though.

sebsx, 2018-01-28:
I can reproduce this on both an original DYS F4 and on a BFF3. The BFF3 has a beeper and I can confirm it goes off continuously and the OSD becomes non responsive.

Any way I can help?

AndersHoglund, 2018-01-28:
OK, not only me then. What versions/revisions have you tried? PR #5032 seems to fix the problem for me. If you have not tried that, please do and report back. Build 552 or later, https://ci.betaflight.tech/job/Betaflight
Thnx.

sebsx, 2018-01-28:
I have the build environment setup on my Mac so I can compile at will if needed, in case you want me to give a certain branch (or even a commit) in your repo a go just say the word.

The one I had lockups with was compiled before your latest PR was merged. I recompiled from master and will flash, test and report back tonight after I get home.

AndersHoglund, 2018-01-29:
OK. Yes, pre-PR5032 versions do have this problem. It would be great to get a report on tests with some post-PR5032 version.

sebsx, 2018-01-29:
Still hangs for me on 870c23875. First I tried HEAD (e3cbff148) which didn't start up properly on the BFF3 (probably the DMA changes).

Managed to make it crash on the 7th try, every time I sent a combination of freq, band, channel and power. Same as before, beeper goes off and everything is unresponsive.

AndersHoglund, 2018-01-29:
OK, thanks for testing. A couple of Qs. 11mS or 22mS frame rate? Using telemetry with a 4649T Rx ??
Yeah, HEAD at thar point seemed DEAD. Rebuilt and test 870c238 on BFF3 too, 25 VTX Setup ops without issues....
Will dig some  more.

AndersHoglund, 2018-01-30:
Rebuilt current HEAD (  dbf08834 ) and after about 15 VTX Setup ops I got a hangup, with beeper constant on. Prior to that, at test 10 or 11, I got this weird beep. Instead of the normal single short config confirmation/notification beep ( `beeperConfirmationBeeps(1)` ) I hear first a tunkated very short beep and then a longer one.  What else is using the beeper at this point? Are all beeper code reentrant and safe??

sebsx, 2018-01-30:
I'm on 11ms, telemetry enabled with SPM4649T, unfortunately I do not have a debug env setup.

Let me know when you want me to test again. Cheers!

AndersHoglund, 2018-01-30:
Ah, same env as me. 11mS and SPM4649T. 
Did some tests using a SPM4648 without TM, same BF version as above. About 50 VTX setup ops without crashing. But I have done that before with TM on also. Somewhat inconclusive. How many tests to have any confidence, 100, 500, 1000? My poor Dx18 will be worn out....
Testing without beeper is also on the agenda.

AndersHoglund, 2018-01-30:
I think beeper behaving is only a symptom. 
Instead I took a look with GDB where we are when crash has occured, by only hitting run/pause/run i gdb:
```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08008cea in srxlRxSendTelemetryDataDispatch ()

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000212 in dispatchProcess ()

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08008cea in srxlRxSendTelemetryDataDispatch ()

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08008cec in srxlRxSendTelemetryDataDispatch ()

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08008cea in srxlRxSendTelemetryDataDispatch ()

```
I can also see there are no TM data to send. So, are the scheduler constantly dispatching srxl TM transmition but srxl has nothing to send? Deadlock? Hmm....

AndersHoglund, 2018-01-30:
Did a couple of tests without confirmation beeps. Still crashes. BUT this also gave the opportunity to catch those spurious beeps I have been hearing, by a GDB breakpoint on the central `beeper()` function.  Seems I get BEEPER_RX_SET beeps. I.e. Tx AUX channel triggered beeps. 
All this is most probably caused by incomplete parsing of the incoming Tx/Rx frames. VTX control bytes might also "leak" as ordinary RC channel data. Will look into this a bit deeper.

AndersHoglund, 2018-01-30:
@sebx, could you please build and try out PR #5071. Thnx.

sebsx, 2018-01-30:
Built and flashed, at how many tries should I stop if it doesn't crash anymore?

AndersHoglund, 2018-01-30:
That is a good Q. A constant question with intermittent problems. I am sure I found and fixed the spurious beeper issue, but if and how that is connected to crashes/hangsups is not really known.... 
I usually give up testing around 50 tries...

sebsx, 2018-01-30:
Done another test to 50, no crashes so this is very encouraging. Well done.

AndersHoglund, 2018-01-31:
Great news, thnx @sebsx . Time to launch this fix live to a larger crowd, that is the real test.

sebsx, 2018-02-01:
Awesome thanks. Let me know whenever you need testing help for the Spektrum stuff.