betaflight/betaflight #4993: motors and buzzer still do not work right after #4973 and #4992

TheIsotopes, 2018-01-20:
@mikeller unfortunately, your fix #4992 for #4973 and #4990 has still not helped properly
The motors begin to turn and hum when i increase the throttle as if they were not synchronized
also i forget too say the buzzer not working any more ... after plugged in the battery the buzzer is good but if the initialisation from fc ended the buzzer seems like inverted and only makes ticking noises

mikeller, 2018-01-20:
@TheIsotopes: Did you do a 'Reset to Defaults' like I mentioned in #4990?

TheIsotopes, 2018-01-20:
Of course, I always flash with full chip erase and then again `defaults` in the cli

TheIsotopes, 2018-01-20:
@mikeller another little update ...
the hum of the motors does not come with throttle input but only on roll stick input ... if i move the copter in the hand the motors makes no humming noise only on roll stick input :(

mikeller, 2018-01-20:
@TheIsotopes: Can I get a `diff` please?

TheIsotopes, 2018-01-20:
you are welcome ... here my `diff` output
```
# Betaflight / MATEKF405 (MKF4) 3.3.0 Jan 20 2018 / 04:20:44 (b579ef270) MSP API: 1.37

# name
name TriHX

# resources
resource MOTOR 1 C08
resource MOTOR 3 C09
resource MOTOR 4 NONE
resource SERVO 1 C06

# mixer
mixer TRI

# servo
servo 5 1150 1850 1550 100 -1

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -OSD
feature LED_STRIP
feature AIRMODE
feature DYNAMIC_FILTER

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER1234

# led
led 0 9,2::CW:6
led 1 9,3::CW:6
led 2 9,4::CW:6
led 3 9,5::CW:6
led 4 6,5::CW:6
led 5 6,4::CW:6
led 6 6,3::CW:6
led 7 6,2::CW:6

# aux
aux 0 0 0 1600 2000 0
aux 1 13 1 1000 1200 0
aux 2 15 0 1000 1200 0
aux 3 26 0 1600 1800 0
aux 4 36 1 1800 2000 0

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rc_interp_ch = RPYT
set max_aux_channels = 2
set serialrx_provider = SUMD
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT300
set tri_unarmed_servo = OFF
set pid_process_denom = 1

# profile
set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 100
set p_roll = 44

# rateprofile
set roll_rc_rate = 145
set pitch_rc_rate = 145
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set roll_srate = 75
set pitch_srate = 75
```

mikeller, 2018-01-21:
@TheIsotopes: Yes it does - now I know how to replicate it, and I have a better idea where to look. Thanks.

richard-scott, 2018-01-21:
Maybe the D Setpoint Weight/Transition are interfering with RC Interpolation or vice versa?
Perhaps try setting D Setpoint Weight/Transition to defaults (weight=0,transition=1) and see if that helps.

mikeller, 2018-01-23:
@TheIsotopes: Can you please test this firmware:

[betaflight_3.3.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1657961/betaflight_3.3.0_MATEKF405.zip)

Not sure if this will fix it, `fc_rc.c` does some things that I am not entirely clear on if they are the way they should be, but using `rc_interp_ch` to influence what channels the rates are applied to seems to be wrong no matter what.

TheIsotopes, 2018-01-24:
@mikeller 
have tested it and everything seems to run normally again when `rc_interp_ch = RPYT` is activated ... thx 👍

mikeller, 2018-01-24:
Thanks for testing @TheIsotopes. I don't have a fix yet, what I did was just a quick proof of concept to test if hardcoding the rates to **not** be applied to throttle would fix it - turns out it does. Fix is forthcoming.

TheIsotopes, 2018-01-24:
if the fix is there i will test it again to confirm ;-)

mikeller, 2018-01-24:
@TheIsotopes: Can you please test this (from #5022):

[betaflight_3.3.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1659867/betaflight_3.3.0_MATEKF405.zip)

TheIsotopes, 2018-01-24:
@mikeller tested now and all is working perfect 👍

PR #5022 (2018-01-25): Fixed rates / interpolation bug.
Fixes #4993.
closed 2018-01-25 by mikeller