betaflight/betaflight #5013: Multiple current sensor support

fredericmorin, 2018-01-22:
I have a high current high voltage application that leads me to use multiple current sensor.

It would be great to add support for multiple ADC based current sensor, as well as ESC sensor and have all of those work at the same time and be visible on the OSD.

I have a big one meter drone with 6 to 8 blheli32 8s 80A esc on it. Will have one current meter per side and blheli32 can also provide current reading.

jflyper, 2018-01-22:
Why multiple ADC based sensors?

jflyper, 2018-01-22:
Oh, you want to place multiple sensors and add them up?

fredericmorin, 2018-01-22:
motors pull 200A total sustained. I'm spreading the load across 2 PDB / current sensor to avoid having to use expensive parts

jflyper, 2018-01-22:
Easiest way would be to handle multiple current ADC inside the ADC driver, add up and scale readings before passing it to the current sensor io module. It would be a custom firmware for your application, though.

fredericmorin, 2018-01-22:
What about adding that to the current code base ? I'd rather see something that can be used by anyone.

mikeller, 2018-01-22:
@jflyper:

> Another way of dealing with multiple current sensor is to create a sensor hub that collects, computes and deliver a single value over a communication link.

The ESC sensor code already does that.

Thinking of this, @fredericmorin, I know it's possibly not the 'cheapest possible' solution for you, but have you considered using ESCs that support ESC telemetry (currently BLHeli32 / KISS), and getting the current consumption as the sum of the consumption of every ESC? This will work out of the box.

fredericmorin, 2018-01-22:
Since I have 6 to 8 motors (not sure yet) wont that require one free uart per esc ? Like... 8 uart + rc + smartaudio ?

mikeller, 2018-01-23:
@fredericmorin: No, the way that ESC telemetry works is that the TX pin of all ESCs are connected to the same RX pin on the flight controller. The flight controller then sends a trigger signal to each ESC in turn, as part of the Dshot frame, getting one after each other to send a telemetry frame. So one UART RX pin is all you need on the flight controller - it should even work with SOFTSERIAL.

fredericmorin, 2018-01-23:
omg. ok nvm, I take back my feature request. Will obviously be of very little use to anyone now that blheli32 provides that ootb.

Thanks @mikeller!

closed 2018-01-23 by fredericmorin