betaflight/betaflight #5015: RX task brings 100% cpu load

TheIsotopes, 2018-01-23:
latest bf 3.3.0 brings 100% cpu load with default settings
```
# version
# Betaflight / MATEKF405 (MKF4) 3.3.0 Jan 23 2018 / 11:55:03 (67523493a) MSP API: 1.37

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      4       1       0    0.5%    0.0%         0
01 - (            PID)   1953      41      17    8.5%    3.8%      2638
   - (           GYRO)   7812
02 - (          ACCEL)    500      12      10    1.1%    1.0%       104
03 - (       ATTITUDE)     49      12       9    0.5%    0.5%        10
04 - (             RX) 100000      10       4  100.5%   40.5%      6044
05 - (         SERIAL)     16     828       1    1.8%    0.5%         3
07 - (BATTERY_VOLTAGE)     25       4       2    0.5%    0.5%         0
08 - (BATTERY_CURRENT)     24       1       0    0.5%    0.0%         0
09 - ( BATTERY_ALERTS)      2       2       1    0.5%    0.5%         0
10 - (         BEEPER)     16       5       1    0.5%    0.5%         0
16 - (      TELEMETRY)     41       3       1    0.5%    0.5%         1
18 - (            OSD)     10     497      27    0.9%    0.5%         9
20 - (            CMS)     10       1       0    0.5%    0.0%         0
21 - (        VTXCTRL)      0       0       0    0.0%    0.0%         0
22 - (        CAMCTRL)      0       0       0    0.0%    0.0%         0
24 - (    ADCINTERNAL)      0       0       0    0.0%    0.0%         0
RX Check Function                   8       2                      2784
Total (excluding SERIAL)                       115.0%   48.3%

# 
```
@mikeller  imho #4982 makes this problem
if i'm use my setting `serialrx_provider = SUMD` the cpu load goes normal

mikeller, 2018-01-23:
Thanks for reporting, @TheIsotopes. FYI: When replicating with a board with nothing but USB connected, I get 100% when running SUMD as well.

TheIsotopes, 2018-01-23:
Yes you are right ... tested on my revolt without soldered rx and sumd brings also 100% cpu load

wind0r, 2018-01-23:
rate/hz 100000 for rx task looks strange high. 
if i remember correctly rate/hz 50 is value for link lost and and rate/hz 100 with sbus

TheIsotopes, 2018-01-23:
@mikeller new update ... if tx on and bound with rx the cpu load is normal ... when i get the rx off the cpu load goes to 100%

mikeller, 2018-01-23:
Thanks for the feedback, @TheIsotopes. I got an idea of what's going wrong now.

Szpero, 2018-01-23:
Confirming.
XSR-E + XJT, F4 REVO.

TX - on after quad -  100% CPU
TX - on before quad - normal cpu usage

Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      4       1       0    0.5%    0.0%         0
01 - (       GYRO/PID)   8000      49      41   39.7%   33.3%      1443
02 - (          ACCEL)    491      12       9    1.0%    0.9%        25
03 - (       ATTITUDE)     49      12      10    0.5%    0.5%         2
04 - (             RX) 111111      12       4  133.8%   44.9%      1449
05 - (         SERIAL)     16     112      10    0.6%    0.5%         1
10 - (         BEEPER)     16       5       1    0.5%    0.5%         0
17 - (       LEDSTRIP)     16     111     103    0.6%    0.6%         9
20 - (            CMS)      9       2       1    0.5%    0.5%         0
21 - (        VTXCTRL)      0       0       0    0.0%    0.0%         0
22 - (        CAMCTRL)      0       0       0    0.0%    0.0%         0
24 - (    ADCINTERNAL)      0       0       0    0.0%    0.0%         0
RX Check Function                   9       2                       654
Total (excluding SERIAL)                       177.1%   81.2%

mikeller, 2018-01-23:
@Szpero: Can you test it this fixes it for you please (built from #5016):

[betaflight_3.3.0_REVO.zip](https://github.com/betaflight/betaflight/files/1658005/betaflight_3.3.0_REVO.zip)

Szpero, 2018-01-24:
@mikeller: So far looks fine. Strange behave gone.

TX - turned off.

Betaflight / REVO (REVO) 3.3.0 Jan 24 2018 / 12:48:19 (d00cbc385) MSP API: 1.37

Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       2       1    0.5%    0.5%         0
01 - (       GYRO/PID)   8000      49      41   39.7%   33.3%      3280
02 - (          ACCEL)    991      16      10    2.0%    1.4%       106
03 - (       ATTITUDE)     99      15      10    0.6%    0.5%        11
**04 - (             RX)     49      35      31    0.6%    0.6%        17**
05 - (         SERIAL)     99     332       1    3.7%    0.5%        10
10 - (         BEEPER)     99       5       1    0.5%    0.5%         1
17 - (       LEDSTRIP)     99     116     109    1.6%    1.5%       121
20 - (            CMS)     59       2       1    0.5%    0.5%         0
21 - (        VTXCTRL)      4       2       0    0.5%    0.0%         0
22 - (        CAMCTRL)      4       1       0    0.5%    0.0%         0
24 - (    ADCINTERNAL)      0       2       0    0.5%    0.0%         0
RX Check Function                   3       1                         0
Total (excluding SERIAL)                        **47.5%   38.8%**

mikeller, 2018-01-24:
Thanks for testing, @Szpero!

PR #5016 (2018-01-24): Fix RX update frequency with no serial input.
Fixes #5015.
closed 2018-01-24 by mikeller