betaflight/betaflight #5017: bf 10.01 cause change in board orientation every time on hookup or exiting cli

yosemitez, 2018-01-23:
bf 10.1
btfl 3.2.4
bff4

every time i connect or exit cli, 10.01 it causes a change in board orientation. i have to reset "zaxis" to correct.

this never happened in 10.0 after a moth of use  and yes board is is installed with arrow pointing forward and 0 in "board and sensor alignment"

program also just started to not load in win10 although icon would load and it showed in task manager. had to go back to 10.0 which works perfect.

McGiverGim, 2018-01-24:
If the firmware is the same, and only change is the configurator, I suspect this issue must go in https://github.com/betaflight/betaflight-configurator/issues

The other day I observed that my North arrow was pointing bad in the configurator, but I ignored it thinking it was a problem of calibration. Maybe it was not.

yosemitez, 2018-01-24:
thanks McGiverGim, will post it over there

the only reason i noticed is because after upgrading to 10.1 i checked and changed everything else but did not check orientation and quad flipped don liftoff

mikeller, 2018-01-24:
See betaflight/betaflight-configurator#884.

closed 2018-01-24 by mikeller