betaflight/betaflight #5018: Configured LED Strip brings crazy jumps in Cycle Time

TheIsotopes, 2018-01-23:
if the ledstrip is configured the cycle time fluctuate between 125-250 ... i can replicate this on matekf405 and revolt target ... if i disable the ledstrip then the fluctuation does not come back ... in order to trigger the fluctuation again i only need to re-enable the ledstrip after fc reboot it takes about 1 minute and the fluctuation is back

jflyper, 2018-01-24:
Quick test revealed it happens on 3.2.x line, too. Seems to have introduced somewhere between 3.2.3 and 3.2.4. (3.2.4 has it, 3.2.3 doesn't).

TheIsotopes, 2018-01-24:
imho it goes even further back ... i was able to reproduce it up to 3.2.0 and that's where it starts ... on the last 3.1.7 build i could not reproduce this fluctuation

jflyper, 2018-01-26:
Problem solved???

TheIsotopes, 2018-01-27:
wrong closed ... sorry ... problem still exists :((

TheIsotopes, 2018-02-10:
little update ... if i'm using fast optimisation function on all three optimisation options in makefile the fluctuation will reduced / eliminated

TheIsotopes, 2018-02-18:
tested today latest 3.3 RC2 ... if i'm using light overclock `set cpu_overclock = 192MHZ` it's have the same efect like fast optimisation above ... fluctuation will eliminated and cpu load goes 1% down

@mikeller 
imo maybe we should adjust the optimization for different mcu ... f1 / f3 mcu with current optimization and f4 / f7 mcu with optimisation like above ... the option `-Ofast` brings a larger firmware file but imo what i have seen so far all supported f4 mcu have at least min 512kB flash ram ... so there still enough space for new features

mikeller, 2018-02-18:
@TheIsotopes: If you've got some more time to test, can you check if modifying `make/source.mk` and removing `drivers/light_ws2811strip.c` and `drivers/light_ws2811strip_hal.c` and / or adding them and `io/ledstrip.c` to `SPEED_OPTIMISED_SRC` brings an improvement for this issue?

TheIsotopes, 2018-02-18:
@mikeller tested two times
- first one with removing `drivers/light_ws2811strip.c` and `drivers/light_ws2811strip_hal.c` from `SIZE_OPTIMISED_SRC` and adding them and `io/ledstrip.c` to `SPEED_OPTIMISED_SRC` ... but no improvement 
- second test only removing `drivers/light_ws2811strip.c` and `drivers/light_ws2811strip_hal.c` from `SIZE_OPTIMISED_SRC` ... results also with no improvement

mikeller, 2018-02-18:
Thanks @TheIsotopes. This is weird, considering your earlier results with different optimisation settings seemed to improve things. Looks like there isn't a quick fix for this.

TheIsotopes, 2018-02-19:
sorry @mikeller but what speaks against f4 / f7 mcu's to optimize only with `-Ofast` option

mikeller, 2018-02-19:
@TheIsotopes: The reason it was done the way it is currently done is that the parts that are size optimised should not matter, since they are (mostly) not used during flight time (like CLI), and that it was thought that having consistent optimisation settings would reduce the likelyhood of compiler related bugs, due to less different settings being used.

Just a thought: Did you do `make <target>_clean` in between doing the tests above?

TheIsotopes, 2018-02-19:
> Just a thought: Did you do make <target>_clean in between doing the tests above?

yes ... deleted `obj` folder after every test

mikeller, 2018-02-22:
@TheIsotopes: Can you test if #5266 brings an improvement.

TheIsotopes, 2018-02-22:
@mikeller sorry but i have to open again, but i can not :((
now i do not understand anything anymore ... tested latest `build #626` and the fluctuations is back
using my test files from later today and all is good ... whats wrong

my build system run's on windows 10 64bit with latest cygwin64 and gcc-arm-none-eabi-7-2017-q4-major-win32 toolchain and a frech cloned master branch from today ... is my build system so great different to the build system from ci.betaflight.tech ... i can not imagine anything else now

mikeller, 2018-02-22:
That is odd indeed. Was it the same configuration? Can you do some comparisons to see if it can be reproduced down to using builds from CI / your local setup? It could also be some race condition that only occurs intermittently.

TheIsotopes, 2018-02-22:
we overlapped ... read my edit ... closed again

closed 2018-02-22 by TheIsotopes