betaflight/betaflight #5020: I2C SCL and SDA Pins configured

nyway, 2018-01-24:
On my target,just for test,MCU is STM32F411CEU6.
I configured the Pin of SCL is PB10,SDA is PB3.
I read datasheet of STM32F411xc,
PB10 can use for I2C2_SCL,alternate is AF4,
PB3 can use for I2C2_SDA,alternate is AF9.
In target.h:
#define I2C2_SCL                PB10        // SCL pad
#define I2C2_SDA                PB3        // SDA pad
And compile target success,but in betaflight configurator,CLI mode -resource list-I2C2 configured  fail,can't see anything.
When I 
#define I2C2_SCL                 PB10        // SCL pad
#define I2C2_SDA                PB11        // SDA pad
CLI mode -resource list,Then i can see I2C2 pins have configured  success.
I have not enough pin for I2C,how can I use for PB10 and PB3?
And if use soft-i2c,could it work?
Thanks a lot.(sorry about my English express..)

jflyper, 2018-01-24:
@nyway
The I2C driver is not capable of handling alternate functions other than AF=4 atm.

I’ll enhance it, probably tonight.

nyway, 2018-01-24:
@jflyper Thank you very much.

jflyper, 2018-01-24:
@nyway 
PR is ready. Since I don't have the board, you have to test it.

nyway, 2018-01-24:
@jflyper  Will test immediately

nyway, 2018-01-24:
@jflyper 
Still fail
betaflight_3.3.0
target.h
#define USE_I2C
#define USE_I2C_DEVICE_2
#define I2C_DEVICE              (I2CDEV_2)
#define I2C2_SCL                PB10        // SCL pad
#define I2C2_SDA                PB3        // SDA pad
#define BARO_I2C_INSTANCE       (I2CDEV_2)

Just same

nyway, 2018-01-24:
@jflyper 
One more place i change:
bus_i2c_stm32f10x.c
#ifdef USE_I2C_DEVICE_2
    {
        .device = I2CDEV_2,
        .reg = I2C2,
        .sclPins = { DEFIO_TAG_E(PB10), DEFIO_TAG_E(PF1) },
        .sdaPins = { _**DEFIO_TAG_E(PB3),**_ DEFIO_TAG_E(PF0) },
        .rcc = RCC_APB1(I2C2),
        .ev_irq = I2C2_EV_IRQn,
        .er_irq = I2C2_ER_IRQn,
    },
#endif
now CLI mode like this:
B03: I2C_SDA 2
B04: MOTOR 1
B05: LED 1
B06: MOTOR 2
B07: MOTOR 3
B08: FREE
B09: FREE
B10: I2C_SCL 2
But my  barometer still not working
I'm not  very familiar in github

jflyper, 2018-01-24:
@nyway 
> I'm not very familiar in github

You have to build your firmware based on the branch:
https://github.com/jflyper/cleanflight/tree/bfdev-i2c-allow-exotic-af

If you have a trouble, publish your branch and give me the URL so that I can pull your tagret.[ch], or paste me these files at slack.

nyway, 2018-01-24:
@jflyper 
Got it.
I update code from your PR.
I2C Pins were configured success.Use CLI Mode can see ：
B03: I2C_SDA 2
B10: I2C_SCL 2
But my barometer(SDO to GND,slave address 0x76) still not working(I2C). I looking for the cause.

jflyper, 2018-01-24:
What does `set baro` say?

nyway, 2018-01-25:
In Betaflight Configurator.**Barometer** turn on,but **Mag** icon gray.
I will try my  compass  HMC5883 to confirm if the bmp280 chip trouble

nyway, 2018-01-25:
@jflyper 
Problem solved.
Thanks for your help,Just missing define "USE_BARO",be defined  as "BARO",stupid myself.
Now Barometer and compass all normal working

nyway, 2018-01-25:
closes #5020

closed 2018-01-25 by nyway

jflyper, 2018-01-25:
Oh, if you have the baro icon turned on, then baro chip has been detected, so is the I2C is working. You can also use CLI command ‘status’ to confirm the detection.

What I meant by “what does set baro say?” was to use CLI ‘set baro’ to see relevant variables.

jflyper, 2018-01-25:
Oh, okay :)

fpvg, 2019-08-28:
sorry im a newbie, im trying to hook up a Beitan BN-880 gps to a Matek f405 -osd obsolete lol. there is no scl or sda pins , can i remap or something to wire up compass for gps. thanks

PR #5021 (2018-01-25): I2C (STM32F4) Allow pins with exotic AF
Fixes: #5020

Pin defs in `i2cHardware_s` was extended to handle I2C pins with exotic AF, available with F411CE (and others).