betaflight/betaflight #5025: SPF3 Won't arm on build #541

watageek, 2018-01-24:
maybe I am just dumb and/or lack of sleep. Be gentle. (dynamic, RX_PPM)

[BTFL_cli_MaRtIaN_BF3.3.0.txt](https://github.com/betaflight/betaflight/files/1660517/BTFL_cli_MaRtIaN_BF3.3.0.txt)

Cpu usage is fixed

jflyper, 2018-01-24:
If you are trying to arm on the bench while connected from configurator, you must turn on motor test switch.

If this doesn't solve your problem, please use other places on the net for support.

closed 2018-01-24 by jflyper

watageek, 2018-01-24:
Nor The arming procedure or the knowledge is the culprit ( or maybe it is!). Build 541 simply won't arm with the switch. Didn't tried on the bench as I prefer to fly with the stable releases. I just wanted to test the newest builds. 

Thank you for the inputs.

Best regards