betaflight/betaflight #5037: ICM20689 Infinite spin

nyway, 2018-01-26:
Gyro &Acc : ICM20689 (SPI)
MCU:STM32F405
Betaflight3.3.0
Problem description：
In directional control(yaw),Infinite spin,really danger,disarm will stop.
Moto were very hot.
After change MPU6000.There is no such problem
We test ICM20602 and betaflight-3.2.4 has same problem too.
I'm not sure if the gyro overflows
Video here :
[https://www.youtube.com/watch?v=eugSeazcm68](video)

richard-scott, 2018-01-26:
Maybe its the same issue as #4990 

After flashing try entering in this to the CLI before setting up your FC again.

> defaults
save

richard-scott, 2018-01-26:
See this [comment ](https://github.com/betaflight/betaflight/issues/4990#issuecomment-359117255) in particular.