betaflight/betaflight #5039: 3.2.4: When set gyro_lpf = experimental, gyro & PID rates shown as 1kHz

ctzsnooze, 2018-01-26:
In the betaflight standalone app at 10.1, if I start with loop times at 8k/8k  gyro_lpf = OFD, and then I select gyro_lpf = experimental in the CLI, the GUI interface changes so that both gyro & PID rates are shown at 1kHz.  The loop time remains 125us and the CLI shows the divider at 1.

In the GUI, if I change back to 8k/8k, on saving the display reverts to 1k/1k.

etracer65, 2018-02-20:
I'm looking into this.

etracer65, 2018-02-20:
@ctzsnooze created https://github.com/betaflight/betaflight/pull/5247 to fix the validation issue that was resetting the Gyro/PID rates when changing gyro_lpf.

Separately I'll be examining extending this experimental 32KHz DLPF mode to other 32K capable gyros as the original experimental change was only for the ICM-20689 gyro.

Lastly, I'm going to open a separate issue to discuss eliminating some if not most of the other gyro_lpf options besides OFF and EXPERIMENTAL.  I can't see the value of the other DLPF cutoffs or why we'd want to go back to running the gyro with a 1KHz refresh rate.  Certainly the low frequency cutoffs are no longer viable - like 5Hz!

ctzsnooze, 2018-02-20:
Thanks eTracer!

I thought I saw older versions of this data sheet in which bits 0 and 1 of GYRO_CONFIG were titled CHOICE_B and used to provide other options.  I cannot find those sheets.  I did find an older sheet for the 6500:

[image]
https://www.invensense.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf

Perhaps I confused the two.  The 6500 seems similar to the ICM206xx gyros.  While it shows 3600Hz for '7', I recall seeing a different number in the early MPU6000 sheets, and it could be that I got the two confused.  

In any case, option 7 is deprecated in current documentation.  And there is no mention in current documentation of the influence of bits 0 and 1 of GYRO_CONFIG for the 6000/6050 family; we cannot guess at it.

No-one has been able to test 'experimental' - i.e. option 7 - because of the bug I found, and no-one has complained about that, apart from me, so there seems to be little or no interest in it.  I only was curious about it from a filter testing perspective since the delay in hardware might then be reduced by almost a millisecond.  But, even if it was a real option, we know from 32k experience that some 'second stage' additional software filtering seems to be required for most quads, which most likely will introduce similar delays.  

So I am not personally troubled if 'experimental' disappears - it's probably best to remove the 'experimental' option altogether. 

I haven't enabled any hardware gyro filtering settings other than the 256 with gyro_lpf = off for a long time.  That's not to say that other people may still use them, but I doubt it.  With all our software filtering options, the significant delays of the supplied hardware filters, and with the impending provision of good 'second stage' filters, it's difficult to see anyone actually needing or using anything other than gyro_lpf = off.  Which could allow removal of that CLI choice altogether.  I guess some broad polling of betaflight users might clarify if anyone needs hardware filtering these days.

For ICM206xx gyros there would be value in selecting between the 256, 3000 and 8000 hz options.  

I agree entirely that the gyro_lpf list should drop the lowest two, unflyable, options (5Hz and 10Hz).

It could then and add 3000 and 8000 at the top (meaning 3281 and 8173 type selection choices in ICM gyros)

I guess if the board was MPU6000 any selection of 256, 3000 or 8000 results in 256, or the list changes to suit the board, whichever is easiest ha ha.

Maybe also the name should change to gyro_hard_lpf or something so people don't confuse it with software filters.

Anyway thanks for looking into this, it would be great to tidy it up and simplify hardware filter selection, especially for the ICM series gyros that are now much more commonplace.

etracer65, 2018-02-20:
Each gyro has a slightly different set of DLPF cutoffs.  The ICM-20608G is similar to the MPU6500 but the cutoffs are different.  Likewise the 20602, 20649 and 20689 have their own charts.

A slight misunderstanding - for 32k mode you get the options of 3000 and 8000 only.

For 8K mode you can choose from any of the green highlighted DLPF_CFG values, but 1-6 will put the gyro in 1KHz refresh instead of 8KHz.  So the only viable options are 0 and 7 (250hz and 3000hz).  0 corresponds to gyro_lpf = NONE and 7 corresponds to gyro_lpf = EXPERIMENTAL.

The 20602, 20608 and 20689 seem to have the same cutoff frequencies and match the 20608G chart above.  

The 20649 is the oddball in that it's a 4000dps part and only has a single 32K DLPF at 12000Hz.

[image]

closed 2018-02-21 by mikeller

ctzsnooze, 2018-02-22:
It seems that we need to revisit the way we set hardware gyro filter values.  The original system didn't anticipate there being a 250 hz hardware filter when 'set gyro_lpf = off'.  And I don't know for sure but it doesn't seem easy to choose 3600 or 8800 in 32k mode.

Since, as @etracer65 noted, hardware filter option of 5 would never be used, maybe the choices list chould be 8k, 3k, 250, 176, 92, 41, 20, 10 going forward.

This would work for any gyro; if the gyro doesn't support >250, and the user selected 3k or 8k, the user's selection could get reset to 250 after the save.  Or the user cannot see 8k or 3k if not available.  Either is OK by me.