betaflight/betaflight #5042: Motolab Cyclone F3 board - enabling LEDstrip with DShot disables motor 4

ctzsnooze, 2018-01-26:
If I enable the LEDstrip while DShot 300 is in use on my Motolab Cyclone F3 boards, Motor 4 stops working.  They have Motor 1 connected to the PPM pin, which is necessary for DShot.

There is no problem running multishot or oneshot with LEDstrip active.

I'd be grateful for a suggestion as to possible resource remapping to fix this!  :-)

Resource listing is as follows:

`resource BEEPER 1 A00
resource MOTOR 1 A07
resource MOTOR 2 A06
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A01
resource MOTOR 6 A02
resource MOTOR 7 A03
resource MOTOR 8 A08
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 B03
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 B04
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 A09
resource I2C_SDA 1 NONE
resource I2C_SDA 2 A10
resource LED 1 B05
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 NONE
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 NONE
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 NONE
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource ESCSERIAL 1 A04
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 A05
resource ADC_RSSI 1 B02
resource ADC_CURR 1 NONE
resource ADC_EXT 1 NONE
mixer QUADX
`

jflyper, 2018-01-26:
@ctzsnooze Can we see the `resource list` output for Dshot and non-Dshot cases?

richard-scott, 2018-01-26:
In the DShot [wiki ](https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol#motolab---motof3-cyclone--tempest) page it says this:

> To allow the use of the LED pin Re-map motor 4 to motor 5 signal out a01, which must also be soldered as such.

jflyper, 2018-01-26:
@ctzsnooze Are you experiencing this with 3.3, or 3.2.x?

3.2.x:
LED Strip (PB8) has a DMA channel collision with motor 4 (DMA 1 Channel 3)

3.3:
LED Strip (PB8) has a DMA channel collision with motors 2 through 4.

Since you have the issue with motor 4 only, I presume you are using 3.2.x.

ctzsnooze, 2018-01-26:
Thanks for your help!  This is with 3.2.x.

Resource list with DShot:

> A00: BEEPER
A01: MOTOR 4
A02: FREE
A03: FREE
A04: FREE
A05: ADC_BATT
A06: MOTOR 2
A07: MOTOR 1
A08: FREE
A09: I2C_SCL 2
A10: I2C_SDA 2
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: MPU_EXTI
B00: MOTOR 3
B01: FREE
B02: FREE
B03: SERIAL_TX 2
B04: FREE
B05: LED 2
B06: FREE
B07: FREE
B08: LED_STRIP
B09: LED 1
B10: FREE
B11: SERIAL_RX 3
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: FREE
C14: FREE
C15: FREE
DMA:
DMA1 Channel 1: MOTOR 1
DMA1 Channel 2: MOTOR 3
DMA1 Channel 3: LED_STRIP
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 2
DMA1 Channel 7: MOTOR 4
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

Resource list with Multishot:
> A00: BEEPER
A01: MOTOR 4
A02: FREE
A03: FREE
A04: FREE
A05: ADC_BATT
A06: MOTOR 2
A07: MOTOR 1
A08: FREE
A09: I2C_SCL 2
A10: I2C_SDA 2
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: MPU_EXTI
B00: MOTOR 3
B01: FREE
B02: FREE
B03: SERIAL_TX 2
B04: FREE
B05: LED 2
B06: FREE
B07: FREE
B08: LED_STRIP
B09: LED 1
B10: FREE
B11: SERIAL_RX 3
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: FREE
C14: FREE
C15: FREE
DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: FREE
DMA1 Channel 3: LED_STRIP
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

Hope this helps narrow down my options.  Thank you again! - c

jflyper, 2018-01-26:
WTH??? No collisions???

jflyper, 2018-01-26:
@ctzsnooze Did you change motor 4 to PA1 as suggested by the wiki???

ctzsnooze, 2018-01-26:
beThere was a need to remap Motor 1 to the PPM pin, hence motor 1 is mapped to A07 and soldered to the PPM signal pin.

When I do a dump all, it shows:

>resource MOTOR 4 B01
resource MOTOR 5 A01

So I think that's how they are physically wired, motor 4 to B01.  I didn't do any remapping or non-default soldering yet; motors 2-4 are as per defaults.  That implies that motor 4 is on the B01 pad (i.e. on the motor 4 pad).

But that is really confusing, because you'll notice that in actual resource lists, for both DShot and Mulitshot, Motor 4 is shown as being sent to A01 (motor 5).  I seems odd that my motor 4 wire is soldered to B01, and it works with Multishot - that seems very confusing to me.

Anyway, since I now only use DShot, looks like I should solder the Motor 4 signal wire to A01 (motor 5).

Should I additionally remap in the CLI:

>Resource MOTOR 4 A01 ?

Since the current resource allocation puts motor4 on A01, this may not be necessary.  I'll test it out and get back to you.  Thanks again - C

jflyper, 2018-01-26:
I recommend you do CLI `defaults` and start configuring on fresh default config.

ctzsnooze, 2018-01-26:
That gives me two things to try.... hang on a moment and I'll get back to you

jflyper, 2018-01-26:
Default first 4 motors are
PA4, PA6, PB0 and PB1, as defined in
https://github.com/betaflight/betaflight/blob/v3.2.x-maintenance/src/main/target/MOTOLAB/target.c

Check it out immediately after `defaults`.

ctzsnooze, 2018-01-26:
Just did defaults on a spare board.  Result of 'dump all' is:

>resource MOTOR 1 A04
resource MOTOR 2 A06
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A01
resource PPM 1 A07
resource LED_STRIP 1 B08

The default resource list command shows, whether DShot is off, or on:

A04: MOTOR 1
A06: MOTOR 2
B00: MOTOR 3
B01: MOTOR 4

With DShot on and the LED strip on, the default motor assignments remain as above, but only motors 2 and 3 get DMA channels.

DMA1 Channel 1: FREE
DMA1 Channel 2: MOTOR 3
DMA1 Channel 3: LED_STRIP
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 2
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

After freeing the PPM pin A07 and assigning that to motor 1, three motors get DMA channels:

>DMA1 Channel 1: MOTOR 1
DMA1 Channel 2: MOTOR 3
DMA1 Channel 3: LED_STRIP
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 2
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

And the resource list shows no change in motor 4 assignment (unlike my problem quad!):

>A07: MOTOR 1
A06: MOTOR 2
B00: MOTOR 3
B01: MOTOR 4
B08: LED_STRIP

OK so now I assign motor 4 to the motor 5 pin A01 explicitly, and the resource list then says:

>A07: MOTOR 1
A06: MOTOR 2
B00: MOTOR 3
A01: MOTOR 4
B08: LED_STRIP

and I get DMA channels for all of them

>DMA1 Channel 1: MOTOR 1
DMA1 Channel 2: MOTOR 3
DMA1 Channel 3: LED_STRIP
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 2
DMA1 Channel 7: MOTOR 4
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

Looks like assigning Motor 4 to Motor 5 and using the Motor 5 pin should do the trick.

ctzsnooze, 2018-01-26:
PS the *betaflight* wiki says "Report that with the Mod and remapping to run DSHOT there is a conflict with DMA for LEDs with Motor #4.
Currently LEDs need to be disabled to run DSHOT."
I'll see if Walter can update it with the info from the DShot wiki.  :-)

ctzsnooze, 2018-01-26:
OK I can confirm that it works fine now.  This can be closed.
The solution for 3.2.X requires two steps:
1.  Paste the following lines into the CLI (don't forget to save)
resource MOTOR 4 A01
resource MOTOR 5 NONE
2.  Solder the signal wire for Motor 4 to the Motor 5 hole.

ctzsnooze, 2018-01-26:
What will need to be done to run DShot and LEDstrip in 3.3.X ?

jflyper, 2018-01-26:
All set?

ctzsnooze, 2018-01-26:
Yep all good on 3.2.4 now - just wondering if there will be a lot I'll have to change with 3.3?

ctzsnooze, 2018-01-26:
Thanks heaps for your help!

jflyper, 2018-01-26:
Happy flying :)

closed 2018-01-26 by jflyper