betaflight/betaflight #5046: SPRACINGF3EVO MAG alignment 180 deg wrong

elepete, 2018-01-27:
In betaflight/src/main/target/SPRACINGF3EVO/target.h
line 82 
#define MAG_AK8963_ALIGN CW90_DEG_FLIP
needs to be changed to: CW270_DEG_FLIP

This was changed time some time ago in INAV. Issue is the Z-axis is inverted and the MAG at roll / pitch drifting heavily due to z-axis inversion.

see:
https://github.com/iNavFlight/inav/issues/590

mikeller, 2018-01-27:
Not sure if this alone will fix it. In the fix for iNav, the way the way the orientation is calculated in the driver is changed as well:

https://github.com/iNavFlight/inav/pull/707/files#diff-b0765a975ceb7d528b0ce88f63dcda40L345

Can you please try this firmware to see if only inverting the compass orientation in the target definition fixes it:

[betaflight_3.3.0_SPRACINGF3EVO.zip](https://github.com/betaflight/betaflight/files/1670485/betaflight_3.3.0_SPRACINGF3EVO.zip)

elepete, 2018-01-27:
@mikeller - thanks, your suspection is correct ;-) and I apologize, was having a dull moment on history...
I remember from Inav there is 1-axis that actually had not the correct sign... 

And your special compile did not fix, as it could not. It actually has a strange mag sensor tab readout that clips all mag to 0 as lowest value (before and after mag calibration).

Thanks again for your look inti this!

mikeller, 2018-01-27:
Hmm, do you think going the same way that iNav has gone, and removing the inversion from the orientation readout in the driver would fix this?

elepete, 2018-01-28:
I have done a local compile to not ask more questions... to be somewhat helpful.
In addition to the modified "target.h" the "compass_ak8963.c" need a changes in lines 333 to 335, remove the negation.
    magData[X] = parseMag(buf + 0, mag->magGain[X]);
    magData[Y] = parseMag(buf + 2, mag->magGain[Y]);
    magData[Z] = parseMag(buf + 4, mag->magGain[Z]);

However I have an issue that your compiled version has as well...
Compass and Accelerometer sensors in the sensors tab do clip at the 0 line with lowest value - strange.
However I think than the behaviour of compass in the F3EVO board is correct - comparing magetic field with accelerometer.

I am using the 'gcc-arm-none-eabi-7-2017-q4-major' version of the ARM compiler downloaded here 'https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads'. Is there a special version needed? I remember that was or is the case for INAV.

mikeller, 2018-01-28:
Thanks for testing this, @elepete.

> However I have an issue that your compiled version has as well...

But does it fix the original issue of the MAG orientation? If so I think we should go ahead with this.

@martinbudden: What's your thoughts on this? The fix was done in iNav (iNavFlight/inav#590), and they are relying on MAG a lot more than we are, so I assume if they fixed it we should apply the fix too.

elepete, 2018-01-28:
Yes - it fixes the issue.
For being on the save side, it could be changed like this to leave other than SPRF3EVO untouched - compiled tested = ok (change in "compass_ak8963.c"):
#if !defined(SPRACINGF3EVO)
    magData[X] = -parseMag(buf + 0, mag->magGain[X]);
    magData[Y] = -parseMag(buf + 2, mag->magGain[Y]);
    magData[Z] = -parseMag(buf + 4, mag->magGain[Z]);
#else
    magData[X] = parseMag(buf + 0, mag->magGain[X]);
    magData[Y] = parseMag(buf + 2, mag->magGain[Y]);
    magData[Z] = parseMag(buf + 4, mag->magGain[Z]);
#endif

Unfortunately there is no definition of MPU9250 in the F3EVO target. I would think at least all MPU9250 will need this.

mikeller, 2018-01-28:
Thanks for testing, @elepete.

> #if !defined(SPRACINGF3EVO)

We've put a lot of effort into getting rid of target dependent code outside of target definitions, and we are committed to keeping them out. If this is a systemic error, then it should be fixed with a fix for all targets (and adjustment of targed specific code where required). If not, it should be fixed in a fix that is part of the definition of the affected target.

elepete, 2018-01-28:
That makes sense - thanks for explaining
Just had a look in the INAV code - the sign is in any case without any depedancy positive. As this is the case for 1 year I would expect no side effects and the "-" can just be deleted, and compass is fixed.

mikeller, 2018-01-28:
@elepete:

Looking at https://github.com/iNavFlight/inav/pull/707/files, this might need an update in targets using this driver as well - which is not a problem.

Do you want to open a pull request for this change?

elepete, 2018-01-29:
Hi - yes I want - however, I think I tried that in INAV as well, but I am not sure how that works on github. Is there a manual / tutorial how to make a pull request?

Other question - is there an answer / solution to the zero clipping of acc & mag of the compiled file?

mikeller, 2018-01-29:
@elepete: No problem, we are always happy to help people who want to contribute. This is a short primer on the workflow: https://guides.github.com/activities/forking/ There are more helpful Guides on  https://guides.github.com/ , have a look at them and don't hesitate to ask questions if something is unclear.

elepete, 2018-02-01:
thanks - was a good learning on how to use git and contribute

closed 2018-02-01 by elepete

mikeller, 2018-02-01:
Thank you for making the effort to learn, and for your contribution, @elepete.