betaflight/betaflight #5049: Motor Life Timer/Counter/Odometer

dthubereng, 2018-01-27:
I'd like to implement a feature that keeps track of the time spent on each of the motors. This is sort of like the odometer on a car or the engine hours on heavy equipment. This could be used to track the time a motor has current/voltage applied to it. Initially I was thinking of an individual counter for each motor, stored in non-volatile memory as seconds, although with 32-bit storage better resolution could be possible; this would depend on the resolution of the scheduled task for implementing this.

Each counter would have to have a reset function, executed manually through BFC. So, this would also affect CLI and MSP.

It could also be useful if this information was somehow available to the OSD: motor with maximum number of time displayed or some such thing.

Over time: Pilots could use this information as an indicator that their motors were getting ready to fail. Motor manufacturers would also have (potentially) a bunch of data about how long their motors are lasting.

So, any ideas where to start something like this? :)

brucesdad13, 2018-01-27:
Would be super handy for brushed :)

mikeller, 2018-01-27:
Not sure if this should be per motor, or if one single armed time counter would be easier to manage.

dthubereng, 2018-01-27:
I thought about that, but do pilots/operators replace all four motors at the same time? I'm not sure I would if just one went out, but then again, perhaps that is the usual thing to do. If so, I would agree that only one timer is necessary.

mikeller, 2018-01-27:
That's a fair consideration. Probably not, but will they want to use CLI to update motor hour counts, or just read the total armed hours and use a scrap of paper to do the tracking?

dthubereng, 2018-01-27:
I think if someone is interested in using this feature, they are probably already used to using CLI. Even so, I think there should be something on one of the tabs (motors?).

What about a menu option accessible through OSD?

This idea came to me precisely because of what @brucesdad13 mentioned, while I was "whoop"ing around my house. These little brushed motors only last 4-6 hours and I don't really want to have to jot down times every 4-5 minutes to track the motors.

mikeller, 2018-01-27:
We could make it configurable to support either combined or individual display. The simplicity of having one number show up in OSD / GUI as 'engine hours' by default, and not requiring users to manage them individually, appeals to me.

Also, for brushed motors (which seem to be one of the target audiences, replacing all four motors is probably the most common course of action. If the brushes / sinter bushings on one motor have gone, the others are probably not far behind.

brucesdad13, 2018-01-27:
The recommendation is often to replace the entire set of brushed motors at the same time. I will confess to not having always done this especially if the motors were quite fresh and only one somehow became damaged. I think for me a single counter in minutes would be perfect.

wind0r, 2018-01-27:
Nice Idea :) Just remember when adding this feature that writing to Flash / emulated "EEPROM" takes "a lot" of time so the data only should get written on Disarm. Sounds like another usecase for a Arm/Disarm Hooksystem as planed in #5048 .. 
A downside is that when the LiPo ejects to fast the data could not be written .. may it even could it lead into a invalide config?

Also STM32F4 Flash "only" is rated for 10k cycles. Every Firmware flash, every config change in configurator, osd, lua reduces this lifetime. 
So there maybe even should be a check that the quad was armed for over $x Seconds. Maybe as a CLI  Parameter which also can be used to dsable the feature completly.

I think Timer Resolution should be in seconds else the data would be to inaccurate (e.g flying 3:31 .. should it add 4 minutes or 3 minutes? ;) )

mikeller, 2018-01-28:
> Also STM32F4 Flash "only" is rated for 10k cycles.

One idea to counter that would be to write the value to the SD card for flight controllers with an SD card slot.

> So there maybe even should be a check that the quad was armed for over $x Seconds.

Good idea. Better to get an approximate value and not wear out the flash.

dthubereng, 2018-01-28:
What if it is something included in the blackbox log, but is somehow maintained between erases? Could it be stored in the same memory? I'm not that familiar with the available storage options yet. Still looking into it.

Definitely think this is something done on arm/disarm and could follow similar (obv. not the same) logic to the auto-disarm delay.

MJ666, 2018-01-29:
I see the flash wearing as critical with such feature. If the flash is dead you can waste the FC or the STM need to be replaced. Smal brushed flight controlers likely have no SDCard or SPI flash.

dthubereng, 2018-01-30:
So some FCs don’t have NV memory external to the uC? :/ What’s the resolution of memory space when you write the STM flash? I have to admit I’m a little outdated with some of the tech.

I wonder though, with 10k cycles, what’s the likelihood of wearing out the flash?

mikeller, 2018-01-30:
> I wonder though, with 10k cycles, what’s the likelihood of wearing out the flash?

If you count every arming / disarming cycle, even including short ones during (bench-) testing, then 10k is probably quite achieveable.

dthubereng, 2018-01-30:
It’s def a legitimate concern. And it doesn’t make sense to limit the feature to only those FCs with external memory. Other than blackbox, are there no other features that store any kind of changing data between power cycles? My initial target would be the BBV2.

dthubereng, 2018-01-31:
After a discussion on Slack I think I understand what I have to work with here. So the latest idea is to implement this on larger targets (F4) that have larger page space for data and use the unused flash data space as a "journal" for engine hours. The page only needs to erased/rewritten on config change and when the engine hours fills up the remaining (?) space. This would allow lots of disarm writes to flash without needing to do an erase. For example, on the F4 with 16K page size, 2K used for config, there would be 14K of available space to keep a log of the time; that's potentially over 3000 disarms (at 32bits resolution). As a further note, this "journal" can be reset (squashed) every time the config is updated.