betaflight/betaflight #5053: BF 3.3 - doesn't recognize AUX 4 switch correctly anymore on OBF4

RipperDrone, 2018-01-28:
3.2 is all good, but in last 2 days' 3.3 builds, in switches scheme, AUX4 seems to be mis-interpreted since even at ~1000 (checked in BFC) PID values are being changed and FC ticks. Should be 'noch changes' instead...
[image]

RipperDrone, 2018-01-28:
strange that even w/ AUX3 centered, PIDs are changing/ticks are active. Only when I crank up AUX3 to max, ticking stops. Remember: All this happens w/ AUX4 around 1000, so should be 'No changes'.
Maybe AUX3/4 are being confused?

mikeller, 2018-01-28:
- can I get a `diff` please?
- can you test to see what happens if you swap AUX3 and AUX4 in your config?

RipperDrone, 2018-01-28:
Sure, here's the diff:

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.0 Jan 27 2018 / 22:46:08 (870c23875) MSP API: 1.37

# reset configuration to default settings
defaults nosave

# name
name Druckbaer

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700 0
aux 1 28 1 900 1300 0
aux 2 2 1 1700 2100 0
aux 3 27 0 1700 2100 0
aux 4 13 0 1700 2100 0

# adjrange
adjrange 0 0 3 900 1025 0 2
adjrange 1 0 3 1025 1075 6 2
adjrange 2 0 3 1075 1125 7 2
adjrange 3 0 3 1125 1175 8 2
adjrange 4 0 3 1175 1225 9 2
adjrange 5 0 3 1225 1300 10 2
adjrange 6 0 3 1300 1375 11 2
adjrange 7 0 3 1375 1425 4 2
adjrange 8 0 3 1425 1475 5 2
adjrange 9 0 3 1475 1525 1 2

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set serialrx_provider = SBUS
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 25
set failsafe_throttle = 1100
set failsafe_kill_switch = ON
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set vbat_max_cell_voltage = 45
set current_meter = VIRTUAL
set vbat_scale = 112
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2049
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 461
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2071
set osd_mah_drawn_pos = 2103
set osd_craft_name_pos = 2059
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON

# profile
profile 0

# profile
profile 1

set p_pitch = 50
set i_pitch = 15
set d_pitch = 34
set p_roll = 44
set i_roll = 58
set d_roll = 70
set p_yaw = 25
set i_yaw = 50
set d_yaw = 40
set p_alt = 55
set i_alt = 40
set d_alt = 50
set p_vel = 75
set i_vel = 0
set p_level = 35
set i_level = 20
set d_level = 0
set gps_pos_p = 45
set gps_posr_p = 14
set gps_posr_i = 33
set gps_posr_d = 50
set gps_nav_p = 0
set gps_nav_i = 55
set gps_nav_d = 30

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 20
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

RipperDrone, 2018-01-28:
Swapping AUX3 & 4: Have just re-versioned back to 3.2.3 which is running fine. If you can't see anything suspicious, I will once more flash back to 3.3 and try again... :-)

jflyper, 2018-01-29:
I have no trouble applying `adjrange` to 3.3 and use AUX4 (rotary) and AUX3 (3-pos) just as intended.
I also have done the same using MSP RX and control stick simulator.

I suggest that you do CLI `defaults` to start with clean defaults, setup receiver and then setup adjustment.

closed 2018-01-29 by jflyper

RipperDrone, 2018-01-29:
Alrighty, will try.