betaflight/betaflight #5054: betaflight F4 fc intermittantly arming - Calib error flagged

Ovasp33d, 2018-01-29:
Hi guys,

Myself and another pilot with the BF F4 board are getting intermittent arming. Brand new boards, only installed just after Christmas.

Can spend up to 10 mins at a time getting it to arm, rebooting through the osd. eventually it calibrates and then its fine for a battery or two and then the problem reoccurs.

I seen another guy on here with a similar issue ( sounded the same but he was dismissed as a singular issue ) and I thought the same thing until I found another pilot in my area with exactly the same issue.

Running a taranis and TBS crossfire micros. Anyone else having this issue. Ive just moved across to BF from KISS so Im not 100% up with the config of these boards.

Thanks

jflyper, 2018-01-29:
@Ovasp33d 
Show us a result of CLI command "status" when the board is trying to arm.

Ovasp33d, 2018-01-30:
Thanks @jflyper ,this was while it wouldn't arm

# diff

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.2.3 Dec 11 2017 / 07:35:40 (cb962eda1) MSP API: 1.36

# name
name Shock One

# resources

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1825 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_hardware = NONE
set acc_trim_pitch = 14
set acc_trim_roll = 2
set baro_hardware = NONE
set fpv_mix_degrees = 45
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 111
set pid_process_denom = 1
set osd_cap_alarm = 1300
set osd_vbat_pos = 2517
set osd_rssi_pos = 503
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2485
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83

# profile
profile 2

# rateprofile
rateprofile 0

set roll_srate = 26
set pitch_srate = 26
set yaw_srate = 50

# status
System Uptime: 78 seconds
Voltage: 167 * 0.1V (4S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 2067, max available config: 16384
CPU:5%, cycle time: 127, GYRO rate: 7874, RX rate: 150, System rate: 10
Arming disable flags: CLI

#

Ovasp33d, 2018-01-30:
Something else very strange that I've found is that I can arm everytime with a flat battery, but swapping to a full battery comes up straight away with the calib error

mikeller, 2018-01-30:
@Ovasp33d: That's a valuable finding. Gyros are often sensitive to ripples in the power supply. It could be that something on your builds produces a lot of noise on the power supply when connected to a full battery. One possible remedy could be to install additional caps, either on the power feed into the flight controller, or on the power supply to the gyro chip (might be hard).

Ovasp33d, 2018-01-30:
Thanks @mikeller it is seeming more and more like a hardware issue than a potential software issue

Ovasp33d, 2018-01-30:
Also the fc wont calibrate/arm with a full battery, but while the full battery is still connected I plug in the PC USB cable and the calibration completes straight away...very strange

AlexYukish, 2018-01-31:
It's problem with design of some Chinese FC like omnibus clones. The problem is in 3.3V supply, that power ups giro. It works bad with 3S voltage. You can power up FC from external 5V BEC

Ovasp33d, 2018-01-31:
@AlexYukish Do you know where this power comes into the gyro on the Betaflight F4 board? I'd like to chase it with the multimeter

AlexYukish, 2018-02-01:
No, I don't know. I have had same issue and solved it with external BEC

Saxin, 2018-02-01:
@Ovasp33d Can you scope the 5V and 3.3V (gyro supply) when powered from Lipo ?

dl7vkj, 2018-02-01:
You can try to set moron_threshold to a higher value. See also #4613 .

Ovasp33d, 2018-02-02:
Ill see if I can get it onto a scope and see whats going on with the voltage. I will try the moron_threshold but another pilot who is experiencing very similar issues (his BF F4 is rebooting when Vbat is too high) tried this without success. Poor input voltage does seem to be a likley suspect though.

Ovasp33d, 2018-02-02:
Also going to try a reflow of the reg and the gyro

Mostro214, 2018-02-28:
I have a similar issue on a cl racing f4 with BF 3.3, quad finish calib after 1 minute and then arm. The same board with 3.2.5 works fine.

mikeller, 2018-03-01:
@Mostro214: See https://github.com/betaflight/betaflight/issues/5054#issuecomment-362420500

stretford-chris, 2018-03-17:
@Mostro214  I am going to try 3.2.5 to see if it fixes the issue on SPRACINGF3

stretford-chris, 2018-03-17:
Ok, tried my settings in 3.2.5, but then the accelerometers won't calibrate.  Back to 3.3 noticed defaults didn't have the gyro calibration issue.  I applied my settings leaving out these lines:-
set gyro_lpf = 188HZ
set gyro_sync_denom = 1
set acc_trim_pitch = -4
set pid_process_denom = 1
And all works well in 3.3 
I am not sure what these lines do or where they came from.
Here is my new working diff output:-
# diff
# Betaflight / SPRACINGF3 (SRF3) 3.3.0 Mar  2 2018 / 04:02:50 (177472b4f) MSP API: 1.37

feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature MOTOR_STOP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
aux 0 0 0 1900 2100 0
aux 1 1 1 1400 1650 0
aux 2 2 1 900 1100 0
aux 3 27 0 1400 1600 0
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1100
set rssi_scale = 30
set blackbox_device = NONE
set min_throttle = 1050
set motor_pwm_protocol = BRUSHED
set motor_pwm_rate = 1000
set failsafe_delay = 10
set failsafe_off_delay = 30
set failsafe_throttle = 1100
set vbat_scale = 75
set ibata_scale = 4000
set deadband = 6
set yaw_deadband = 6
profile 0

rateprofile 0

set roll_rc_rate = 60
set pitch_rc_rate = 60
set yaw_rc_rate = 60
set roll_expo = 20
set pitch_expo = 20
set roll_srate = 72
set pitch_srate = 72

alancastell, 2018-07-11:
Was there a solution to this?  I am having this happen on a BFF4 board now as well.
No hard crashes.