betaflight/betaflight #5057: all my FC hanging on

TheIsotopes, 2018-01-29:
@blckmn @mikeller after #5045 all my fc (MatekF405, SPRacingF3EVO, Revolt) hanging on after flash or enabling dshot

blckmn, 2018-01-29:
Looks like it’s hardfaulting. Can you full chip erase? What do you mean by ‘flash’? I’m assuming it flashes ok if you’re turning on dshot?

TheIsotopes, 2018-01-29:
I'm flashing with `Full Chip Erase` ... after flashing and restart the f3evo hangs immediately ... the MatekF405 hanging after enabling dshot

blckmn, 2018-01-29:
Ok thanks I’ll check it out.

mikeller, 2018-01-29:
Additional info to facilitate debugging: BEEROTORF4 is wedging as well after #5045 (with full reset): Boot blinks work, USB device is configured, but LED0 is on solid.

jflyper, 2018-01-29:
Isn't DMA1 ST0 continuously interrupting?

Whenever I hit break under gdb, it stops at `DEFINE_DMA_IRQ_HANDLER(1, 0, DMA1_ST0_HANDLER`.

blckmn, 2018-01-29:
@jflyper the cue I needed. Fix for the DSHOT issue is here : https://github.com/betaflight/betaflight/pull/5059

jflyper, 2018-01-29:
👍

TheIsotopes, 2018-01-30:
@blckmn tested latest build `#559` with merged #5059 and #5060 ... on Revolt and SPRacingF3Evo all working again but the MatekF405 can not read sdcard :((

blckmn, 2018-01-30:
@TheIsotopes can you please test by turning off DMA on the SDCARD? There is an option in the cli.

TheIsotopes, 2018-01-30:
@blckmn it is already off by default

TheIsotopes, 2018-02-01:
@blckmn little update ... tested backward on BF 3.2.4 and latest build BF 3.2.5 ... sd card slot working without problems on this builds ... back to BF 3.3.0 build #555 (and above) and the sd card slot is no more working

TheIsotopes, 2018-02-04:
@blckmn tested now #5121 and it's fixed ... thx 👍

closed 2018-02-04 by TheIsotopes