betaflight/betaflight #5058: Feature Request: Add arm state to telemetry feedback

AlessandroAU, 2018-01-29:
It would be awesome if the arm state would sent back as a telemetry variable, this way the radio would know if the quad was actually armed.

What I would like is to hit the arm switch and when the FC has actually armed it can be relayed through telemetry and the radio will know and can play a track etc.

SwiftingSpeed, 2018-01-30:
In Smartport telemetry, the arming state and several flight mode variables are already sent via the temperature 1 telemetry. The right most digit is the arming state and has three values:
1: Arming disabled (too large angle for example)
2: Arming enabled (but not armed)
4: Armed
I'm not sure but I think maybe they can be combined as well so that 6 is arming enabled and armed.

You can create a lua script to check the right most digit and see if it's less than 4 for not armed and equal or greater than 4 for armed.

varkokonyi, 2018-01-31:
How does "arming enabled and armed" differ from armed?

mikeller, 2018-01-31:
@SwiftingSpeed:

> I'm not sure but I think maybe they can be combined as well so that 6 is arming enabled and armed.

Not in a reliable way - some arming disable conditions are only tracked when disarmed.

SwiftingSpeed, 2018-01-31:
What I meant was that you don't make a script that checks for 4, it's actually (depending on the disarming conditions) 5 or 6 when armed.

It also seems that I have the 1 and 2 flipped in the previous post.
1 indicate that arming is _not_ disabled
If it's not 1 then it's 2

4 is then added if armed so the result is either 5 or 6 when armed since the first check always results in 1 or 2.

1 and 2 can't be combined.

closed 2018-07-10 by stale[bot]