betaflight/betaflight #5063: Please restore the "level_sensitivity".

recrime, 2018-01-30:
Hello,
I and my daughter are angle flyer - Line of sight.

There was a level_sensitivity which was provided as a parameter from betaflight 3.1.0.
- I don't know since when this effect was exist.
https://github.com/betaflight/betaflight/wiki/BetaFlight-V3.1.x
Betaflight 3.1.0-Release

As a angle flyer (LOS), we're using both level_limit and level_sensitivity as below.
set level_sensitivity = 100
set level_limit = 70

This means if I move stick position around between 70~100%, it effects full degree (=level_limit).
Without the level_sensitivity, I should do full stick movement to effect full degree.
As I know, most of toy (closed program?) quads with angle mode show similar behavior.
-If you have 1 cell drone, please test it, it will show full degree before stick reach the full position.

Anyway, the level_sensitivity was removed by below issue.
https://github.com/betaflight/betaflight/issues/3217
https://github.com/betaflight/betaflight/pull/4013

I know "D setpoint weight" which was mentioned in.
https://github.com/betaflight/betaflight/issues/4386
We've tried it, but the quad shows different movement - little similar.

If the level_sensitivity causes some confusion, I think below logic could solve it.
- Logic : If the level_limit is greater than level_sensitivity, betaflight code simply ignore level_sensitivity or forcely set it to level_limit value, 

My daughter and I are players of some drone sport.
Left is my daughter.
https://www.youtube.com/watch?v=Si9MCI8L-pk

We hope that the level_sensitivity is back, please HELP us.

Below is dump of my config which has the level_sensitivity and turtle mode (with spin 4 motors).
We can't use latest version which has turtle mode (with spin 2 motors), as the latest version doesn't have the level_sensitivity parameter.

Thanks.
-----------

I've attached two configuration.
First one has level_sensitivity, but second one has no level_sensitivity.

https://github.com/furiousfpv/betaflight/releases
betaflight-3.2.0-20170730(6268af2)_with level_sensitivity-from furiousfpv_diff all

https://github.com/betaflight/betaflight/releases
betaflight-3.2.0-20170928(ceb6f70)_no level_sensitivity-from betaflight_diff all

'''
diff all

# version
# Betaflight / FF_FORTINIF4 (FORT) 3.2.0 Jul 30 2017 / 20:17:29 (5b31ca2)

# reset configuration to default settings
defaults

# name

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature ANTI_GRAVITY

# beeper

# map
map TAER1234

# serial

# led
led 0 4,15::A:0
led 1 5,15::A:0
led 2 6,15::A:0
led 3 7,15::L:0
led 4 8,15::L:0
led 5 9,15::A:0
led 6 10,15::A:0
led 7 11,15::A:0

# color

# mode_color
mode_color 6 1 12

# aux
aux 0 0 0 1625 2000
aux 1 1 2 1625 2000
aux 2 2 2 1375 1625
aux 3 13 1 1350 1600
aux 4 35 1 1000 1350

# adjrange

# rxrange
rxrange 0 1840 1160
rxrange 1 1160 1840
rxrange 2 1840 1160
rxrange 3 1160 1840

# vtx

# rxfail

# master
set min_check = 1050
set max_check = 1950
set serialrx_provider = SPEK2048
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set small_angle = 180
set auto_disarm_delay = 0
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2

# profile
profile 0

set p_roll = 58
set i_roll = 50
set d_roll = 35
set p_level = 58
set level_sensitivity = 104
set level_limit = 73

# profile
profile 1

set p_roll = 58
set i_roll = 50
set d_roll = 35
set p_level = 58
set level_sensitivity = 100
set level_limit = 70

# profile
profile 2

set p_roll = 58
set i_roll = 50
set d_roll = 35
set p_level = 58
set level_sensitivity = 93
set level_limit = 65

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
'''

'''
# diff all

# version
# Betaflight / FF_FORTINIF4 (FORT) 3.2.0 Sep 28 2017 / 16:23:27 (ceb6f70) MSP API: 1.36

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature ANTI_GRAVITY

# beeper

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1625 2000
aux 1 1 2 1625 2000
aux 2 2 2 1375 1625
aux 3 13 1 1350 1600
aux 4 35 1 1000 1350

# adjrange

# rxrange
rxrange 0 1840 1160
rxrange 1 1160 1840
rxrange 2 1840 1160
rxrange 3 1160 1840

# vtx

# rxfail

# master
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set small_angle = 180
set auto_disarm_delay = 0
set deadband = 3
set yaw_deadband = 3

# profile
profile 0

set setpoint_relax_ratio = 80
set dterm_setpoint_weight = 220
set p_roll = 58
set i_roll = 50
set d_roll = 35
set p_level = 58
set level_limit = 80

# profile
profile 1

set setpoint_relax_ratio = 70
set dterm_setpoint_weight = 220
set p_roll = 58
set i_roll = 50
set d_roll = 35
set p_level = 58
set level_limit = 75

# profile
profile 2

set setpoint_relax_ratio = 85
set dterm_setpoint_weight = 150
set p_roll = 58
set i_roll = 50
set d_roll = 35
set p_level = 58
set level_limit = 70

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
'''

mikeller, 2018-01-30:
Hi @recrime. If you have a look at https://github.com/betaflight/betaflight/issues/3217#issuecomment-325929202, you will see the reason why `level_sensitivity` was removed: It never actually had the desired effect. Just use `level_limit` instead.

Have a look at https://github.com/betaflight/betaflight/issues/4386#issuecomment-337992786 for hints on how to recreate the old, flawed behaviour.

recrime, 2018-01-30:
Hi mikeller,
I already reviewed #3217 and #4386 and mentioned them on above.
I've tried "D setpoint weight and transition" which were described in #4386, it just shows little similar behavior, but actually different.

From Betaflight 3.1.0-Release
- Added new level sensitivity and level limit parameters in degrees. level_limit is the maximum allowed angle. Level_sensitivity is the max deflection on full stick @borisbstyle

The level_sensitivity has different purpose with level_limit.

In ACRO mode, the neutral stick position let quad retain same angle.
But, in ANGLE mode, pilot should keep stick position for retaining same angle.

Yes I could increase the angle_limit without the level_sensitivity, but it's not easy move full stick position or retain stick position on aileron and elevator for same angle. I hope that the level_sensitivity is back for precise and aggressive control on angle mode.

Let me explain with some settings.
1) set level_limit = 70, with no level_sensitivity
   I should do full stick position for 70 degree of pitch, roll.
   If I need mix elevator and aileron together, it's hard to control.
2) set level_limit = 90, with no level_sensitivity.
  With 77.8% of stick position, the quad could be reached 70 degree on pitch, roll.
  But, retaining 70 degree (=77.8% stick position) is very difficult. 
  If I need mix elevator and aileron together, it would be more difficult.
3) set level_limit = 70, with level_sensitivity =100
  With 70% or above of stick position, the quad could be reached and retained 70 degree.
  And when I need mix elevator and aileron, I could control the quad more precisely and aggressively.

Thanks for your help

mikeller, 2018-01-30:
@recrime: Ah, I think now I understand. What you actually want is a deadband around the full stick positions, so that once you have reached a certain deflection, the angle will not change even when you move the sticks?

recrime, 2018-01-30:
@mikeller Yes, like a deadband between at 70~100% stick position. I think it's very important for the aggressive ANGLE flyer. It also very helpful for young person - like my daughter, or for small hands people. :)
But, for ACRO flyer, it's meaningless.

So I hope that the level_sensitivity is back.

For the previous issue #3217, I'm suggesting below logic.
- Logic : If the level_limit is greater than level_sensitivity, betaflight code simply ignore level_sensitivity or forcely set it to level_limit value.

Thanks for your help.

mikeller, 2018-01-30:
@recrime: While the way you were using this makes sense now that you have explained it, the way it used to have to be configured does not - after all it was removed because it was not understandable what this could be used for - and according to @borisbstyle's comment it is also not what it was intended for.
For these reasons I would suggest that we try to work out a better way to configure this, which will also make it possible for new flyers to configure and use this without having to be told first how to set this up.

How about keeping `level_limit` as it is now, and introducing `angle_upper_deadband` that allows to explicitly set a deadband at the top of the stick range for ANGLE mode?

recrime, 2018-01-30:
@mikeller : Sorry, I'm not a developer, so I can't fully understand "angle_upper_deadband" will do.

My understanding is.... something like below?
 1) Default : level_limit=55(degree), angle_upper_deadband=0(%)
    -> At 50% stick position, angle is 27.5 degrees.
        At 100% stick position, angle is 55 degrees.

 2) level_limit=50(degree), angle_upper_deadband=10(%)
    -> At 45% stick position, angle is 25 degrees.
        At 90%~100% stick position, angle is 50 degrees.

 3) level_limit=70(degree), angle_upper_deadband=30(%)
    -> At 30% stick position, angle is 30 degrees.
        At 50% stick position, angle is 50 degrees.
        At 70~100% stick position, angle is 70 degrees.
    -> This is same with level_limit=70 and level_sensitivity=100.

If my understanding is correct, I thinks it's ok and could remove confusion of level_sensitivity.

mikeller, 2018-01-30:
@recrime: Yes, that is how I thought this should work. Renamed this issue to reflect the new goal.

recrime, 2018-01-30:
@mikeller : I really appreciate your help.

martinbudden, 2018-01-30:
@recrime , @mikeller , am I missing something, or could this not be achieved by adjusting the rates?

mikeller, 2018-01-30:
@martinbudden: Not sure I understand this right. But as far as I understand, `level_limit` defines the maximum commanded angle of the craft at full stick throw. Not sure how rates could be used to configure it such that this angle is reached at a throw less than full stick throw, unless you are referring to rates on the TX, which will make this hard to configure, and susceptible to limiting on the TX / RX hardware that is used.

mikeller, 2018-06-03:
@recrime: Have a look at #5970. I think this can provide a 'trainer' feature for acro mode that is even more powerful than the deadband that you are proposing.

recrime, 2018-06-04:
@mikeller : Thanks for your sharing information and effort. But, it seems that it's different with the deadband feature. The deadband feature is needed for angle flight like below video and leveling (angle mode) is needed for quick recovery after bumping against obstacles.

https://youtu.be/cuueDN8iVTU

She is using ;
  frame : qx90c
  betaflight : 3.1.6
     angle mode
     level limit : 63
     level sensitivity : 80

closed 2022-06-03 by haslinghuis