betaflight/betaflight #5074: Gps not initializing

donalfenwick, 2018-01-30:
I've encountered an issue where after upgrading betaflight where my GPS unit was no longer detected when the quad powered up. Flight controller is an omnibus F4 V5 and Gps unit is [HGLRC Neo-M8N M8M Ublox 8](http://www.hobbyrc.co.uk/hglrc-neo-m8n-m8m-ublox-8-tiny-72-channel-gps-and-glonass-receiver-module-for-inav)

After some trial and error I noticed that if I connected the GPS unit after the FC powers up then it seems to detect it and the num sats icon appears in the OSD. If the GPS is connected before boot then the OSD element never appears.

Issue seems to have been introduced in BF 3.2.3 I've narrowed it down the change in #4666
Reverting this bug fix on master seems to get the GPS unit working again.

I'm not really sure about how the baudRateIndex is configured to suggest a fix here.

mikeller, 2018-01-30:
@donalfenwick: Can we have a `diff` please?

jflyper, 2018-01-30:
WTH???

donalfenwick, 2018-01-30:
@mikeller Sure, no problem.

```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.3 Dec 11 2017 / 07:56:44 (cb962eda1) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature GPS
feature TELEMETRY

# beeper

# map
map TAER1234

# serial
serial 0 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = CRSF
set gps_provider = UBLOX
set osd_vbat_pos = 44
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 2129
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = 1

# profile
profile 0

# rateprofile
rateprofile 0
```

mikeller, 2018-01-31:
This is weird, the config is looking ok.

@jflyper: Could it be that we are reading back the speed value somehwere, and then expecting the speed instead of table index the second time round?

jflyper, 2018-01-31:
I wanted to test this tonight, but it looks like my puck is blown by some reason...

@mikeller I reviewed the code itself, and it is flawless in terms of baudrate parameter to `openSerialPort` and `setBaudrate`.

@donalfenwick You are using auto baud = OFF, auto config = ON and fixed baudrate at 115200, is this correct?

donalfenwick, 2018-01-31:
@jflyper Yes, I can confirm thats how its configured. 
I set the baud rate to 115200 with the ublox u-center tool and confirmed it was working at that rate before I hooked it up to the quad. Hadn't actually though to try setting it to auto baud to see if that makes any difference. I'll give it a go this evening.

McGiverGim, 2018-01-31:
I'm using a Ublox M8N with auto-config and auto-baud and it's working. I don't know if this can help.

Maybe the auto-config is changing the baud rate? If that's true, I'm sure it will not be 115200.

donalfenwick, 2018-01-31:
Ok, so if I set the baud rate to auto or anything less than 115200 on the ports tab of the configurator it seems to work fine.
If I set it to 115200 (which I had configured the unit for with ublox) it doesn't work until I physically disconnect and reconnect the GPS unit. Then it appears in the OSD and starts picking up sats.

I guess before the baudRateIndex bug was fixed it was probably just selecting the lowest baud rate from the enum and that rate worked fine.
I'm thinking that its maybe just an issue with the GPS unit itself when running at 115200 then. Just not sure why it would work with BF set to 115200 in the disconnect connect scenario. Probably just a dodgy GPS I guess.

donalfenwick, 2018-01-31:
Thats probably it. Thanks guys!

closed 2018-01-31 by donalfenwick