betaflight/betaflight #5084: feature request:  using free i/o Pin on the STM32 chip or resource mapping to shutdown the VTX

bnn1044, 2018-01-31:
Hi: I am trying to look into the code to get a free I/O pin to turn on and off the VTX power. people been telling me it is very useful feature for the team racing on pit stop. even on big race.
1. power up your quad, the VTX is not power up. untill you use a radio Switch to turn the VTX power on.

hardware side.
1. like some FC like CLRACINGF4, it has a dedicated VTX and camera power BEC, and also this BEC has a Enable pin. so it is very easy to implement this function. for those FC planning to use the Vbat power to power the VTX and camera, also can be done using a small mosfet through the I/O pin from the MCU.
also when the quad is Armed, need to disable this function, only work when the quad is not arm.

2. if anyone can gen up the code for me. I can make the change on the CLRACINGF4 to implement this function. will be great.

jflyper, 2018-01-31:
A simple BOX to GPO (General Purpose Output) is relatively easy, but it should be implemented as a general purpose output, not only for vtx power switching as there are other possible use cases, such as landing lights.

ledvinap, 2018-01-31:
ARM to GPIO is another use case ...

jflyper, 2018-01-31:
We have a PR for that already. Do you think it should be consolidated?

ledvinap, 2018-01-31:
Both use cases can be in periodic task (~20Hz, or beeper/leds) that will copy state according to configuration ... 
Then use permanentBoxId as source identification, ioTag_t as output pin, allow few (~4) instances.

bnn1044, 2018-02-01:
or someone can point me a direction how to start?

mikeller, 2018-02-01:
@bnn1044: Have a look at #5064. This pull request does something very similar to what you are proposing, so you can use it as a template.

jflyper, 2018-02-02:
@bnn1044 
There is a PR #5108 and #5109.
You can pull #5109 to start experimenting.
As soon as you configure a pin with `resource pin <1-4> <pinid>`, corresponding box will be available for configuration in mode tab.

bnn1044, 2018-02-02:
@jflyper @mikeller Will do. thanks.

brucesdad13, 2018-02-03:
Great idea! :)

jflyper, 2018-02-03:
@bnn1044 FYI, there is another PoC #5110, but you can continue your experiment with #5109; the new one provides wider mapping capability, but doesn't affect your application.

totterfree, 2018-03-12:
I'm also very interested in this feature. SA is simply not robust enough for team racing. I think a much cleaner solution would be for VTXs to better control their output in regards to both frequency and power. If the protocol with which we controlled said VTXs was open source and well defined, we would see greater advancements in this area. Until that happens, I think this is really good concept which would provide a lot of value for people who participate in team racing.