betaflight/betaflight #5085: Airmode throttle-level activation not correct for 3D mode

etracer65, 2018-01-31:
The net result is that in 3D mode airmode is activated immediately on arming and there is no i-term windup prevention.  Looks like it's been this way ever since the airmode activation throttle level was added (basically forever).

In `fc/fc_core.c` in the function `processRx()` the logic only looks at the raw throttle value (in microseconds) being above `airmode_start_throttle` (default 1350).

```
    if (isAirmodeActive() && ARMING_FLAG(ARMED)) {
        if (rcData[THROTTLE] >= rxConfig()->airModeActivateThreshold) {
            airmodeIsActivated = true; // Prevent Iterm from being reset
        }
    } else {
        airmodeIsActivated = false;
    }

    /* In airmode Iterm should be prevented to grow when Low thottle and Roll + Pitch Centered.
     This is needed to prevent Iterm winding on the ground, but keep full stabilisation on 0 throttle while in air */
    if (throttleStatus == THROTTLE_LOW && !airmodeIsActivated) {
        pidResetITerm();
        if (currentPidProfile->pidAtMinThrottle)
            pidStabilisationState(PID_STABILISATION_ON);
        else
            pidStabilisationState(PID_STABILISATION_OFF);
    } else {
        pidStabilisationState(PID_STABILISATION_ON);
    }
```

In 3D mode the throttle has to be centered (around 1500) to arm so airmode will immediately activate preventing the i-term windup suppression from happening.

As a proposed fix the airmode activation throttle should be changed from an absolute microsecond value to a throttle percentage.  Then the new function `calculateThrottlePercent()` (which is 3D aware) added as part of Runaway Takeoff Prevention can be used for the airmode activation logic.

The only problem is that `calculateThrottlePercent()` currently returns an integer percentage and the equivalent to the current airmode activation throttle of 1350 would be 13.5%.  So we'd either have to change the `calculateThrottlePercent()` function to return a float (and store a float as the parameter value), change the function result and parameter value to be scaled integer (like 135 for 13.5% - possibly confusing), or just change the default to 13%.

mikeller, 2018-01-31:
Good find, @etracer65. Shows how little testing we have for airmode. :-| Please go ahead with a pull request.

etracer65, 2018-01-31:
@mikeller Ideally I think we should replace `airmode_start_throttle` (uint16_t) with `airmode_start_throttle_percent` (uint8_t).  It looks straightforward except for I'm not sure what to do with interface/msp.c.  It seems the `airModeActivateThreshold` (internal name) is being sent to/from the configurator via MSP.  Not sure entirely why as it's not exposed as a configuration item.  Maybe it's part of the profile and needs to be there to copy profiles?
```
    case MSP_RX_CONFIG:
        sbufWriteU8(dst, rxConfig()->serialrx_provider);
        sbufWriteU16(dst, rxConfig()->maxcheck);
        sbufWriteU16(dst, rxConfig()->midrc);
        sbufWriteU16(dst, rxConfig()->mincheck);
        sbufWriteU8(dst, rxConfig()->spektrum_sat_bind);
        sbufWriteU16(dst, rxConfig()->rx_min_usec);
        sbufWriteU16(dst, rxConfig()->rx_max_usec);
        sbufWriteU8(dst, rxConfig()->rcInterpolation);
        sbufWriteU8(dst, rxConfig()->rcInterpolationInterval);
        sbufWriteU16(dst, rxConfig()->airModeActivateThreshold);
        sbufWriteU8(dst, rxConfig()->rx_spi_protocol);
        sbufWriteU32(dst, rxConfig()->rx_spi_id);
        sbufWriteU8(dst, rxConfig()->rx_spi_rf_channel_count);
        sbufWriteU8(dst, rxConfig()->fpvCamAngleDegrees);
        break;
```
and
```
    case MSP_SET_RX_CONFIG:
        rxConfigMutable()->serialrx_provider = sbufReadU8(src);
        rxConfigMutable()->maxcheck = sbufReadU16(src);
        rxConfigMutable()->midrc = sbufReadU16(src);
        rxConfigMutable()->mincheck = sbufReadU16(src);
        rxConfigMutable()->spektrum_sat_bind = sbufReadU8(src);
        if (sbufBytesRemaining(src) >= 4) {
            rxConfigMutable()->rx_min_usec = sbufReadU16(src);
            rxConfigMutable()->rx_max_usec = sbufReadU16(src);
        }
        if (sbufBytesRemaining(src) >= 4) {
            rxConfigMutable()->rcInterpolation = sbufReadU8(src);
            rxConfigMutable()->rcInterpolationInterval = sbufReadU8(src);
            rxConfigMutable()->airModeActivateThreshold = sbufReadU16(src);
        }
        if (sbufBytesRemaining(src) >= 6) {
            rxConfigMutable()->rx_spi_protocol = sbufReadU8(src);
            rxConfigMutable()->rx_spi_id = sbufReadU32(src);
            rxConfigMutable()->rx_spi_rf_channel_count = sbufReadU8(src);
        }
        if (sbufBytesRemaining(src) >= 1) {
            rxConfigMutable()->fpvCamAngleDegrees = sbufReadU8(src);
        }
        break;
```
So changing the data type seems kind of messy and may require configurator changes?

Basically I think shifting to a simple integer percentage defaulting to 13 would be adequate as this ends up being really close to the previous 1350 value when considering the min_check deadzone.

mikeller, 2018-01-31:
From the look of things, `airModeActivateThreshold` was added to MSP, but it was never used in the configurator:

```
mikeller@pc:~/git/betaflight-configurator$ grep -r airModeActivateThreshold .
./src/js/msp/MSPHelper.js:                    RX_CONFIG.airModeActivateThreshold = data.readU16();
./src/js/msp/MSPHelper.js:                    RX_CONFIG.airModeActivateThreshold = 0;
./src/js/msp/MSPHelper.js:                    .push16(RX_CONFIG.airModeActivateThreshold);
./src/js/fc.js:            airModeActivateThreshold:   0,
mikeller@pc:~/git/betaflight-configurator$ 
```

But we don't know if other applications that are using MSP are using this value. So the best thing to do is to preserve backwards compatibility, and scale `airmode_start_throttle_percent` by a factor of 10 when sending / receiving over MSP.

etracer65, 2018-01-31:
Looks like it's going to be messy no matter what.  Storing it as 135 to represent 13.5% and passing it as 1350 via MSP.

Maybe it's best to leave the value alone and just silently scale 1350 to 13.5% when checking the throttle.  This is not exactly a parameter that many people even know about let alone tune so it might be better just to fix it behind the scenes even though it's not ideal.

I would still probably change the default to 1300 because the `calculateThrottlePercent()` function takes into account min_check and 3D deadzones so raw 1350 microseconds is really actually about 13% real throttle in default configurations.

mikeller, 2018-01-31:
What I had in mind was to store it as 13 (i.e. in percent) and then scale the value to be sent over MSP up to 1300 (that makes it a factor of 100 :wink: ), and do the reverse (i.e. convert 1350 received over MSP down to 13) when it is being set over MSP.

etracer65, 2018-01-31:
I kind of got that but my programmer hind-brain wasn't thrilled by the precision loss - not that it really matters though.  1300 - 1399 = 13% seems good enough.  Might be really confusing though for whatever (if anything) uses this parameter over MSP.  The user configures 1340 and it keeps resetting to 1300.

etracer65, 2018-02-01:
After the brain-fart and realizing that 1350us is really 35% throttle (thanks @mikeller ), all the confusion is resolved and https://github.com/betaflight/betaflight/pull/5092 has been submitted.

etracer65, 2018-02-02:
Fixed by https://github.com/betaflight/betaflight/pull/5092

closed 2018-02-02 by etracer65