betaflight/betaflight #5086: "Smartaudio" for akk/rdq vtx

ijustwannafly, 2018-01-31:
I have read on different locations that the smartaudio "bug" that bf introduced has been fixed to be up to spec to tbs.

I use the mach2 vtx but seeing as how akk is the manufacturer (from what I gather) it doesn't look like there will be any easy or available way to update the code on the vtx itself.

Akk just posted a github link with the changes made to keep the "bug".

Would it be possible to allow this so that the vtx's that were purchased (that may not be fixed via vtx firmware) can still operate?

https://github.com/krankenreif/betaflight/commit/90efdec9da7722a74121758379a004dec38e9146

If this is a definite no then I will close this inquiry.

Thank you.

mikeller, 2018-01-31:
Since it is AKK's implementation of SmartAudio that does not comply with the specification, it is up to them to either provide fixed VTX firmware, or replace the faulty devices under warranty.

ijustwannafly, 2018-01-31:
Thanks mikeller.  I figured as much but when there isn't much info on if the vtx itself would be fixed via firmware on the device itself it would seem that warranty replacements may not happen.

Thanks for the insight.

closed 2018-01-31 by ijustwannafly

mikeller, 2018-01-31:
@ijustwannafly: In some jurisdictions (thinking of EU but not sure), consumer protection laws say that suppliers have to give warranty in the case of goods not being 'fit for purpose', even if this is not backed by the manufacturer. To me, this looks very much like a case of 'not fit for purpose'.

ijustwannafly, 2018-01-31:
@mikeller 

Thanks for that info.  Sure would be nice if I could just use the swd rst pads on the hardware to flash the legit protocol like we users can with flight controllers that have them (swd ports).

mikeller, 2018-01-31:
@ijustwannafly: There are other manufacturers (like Matek) that have VTX with firmware update capability on the market, and use it to fix bugs in the firmware: http://www.mateksys.com/?portfolio=vtx-hv#tab-id-6

ijustwannafly, 2018-01-31:
@mikeller 

Agreed.  In fact I have been eyeing the furiousfpv stealth long range with tramp telemetry but the price tag is hard to swallow initially.  However, here's hoping the openvtx project is fruitful AND it can be flashed to any vtx existing or new.

ijustwannafly, 2018-02-02:
@mikeller,

Digging around some more the hope is largely bleak.

I guess I found some information stating that all future smartaudio devices have to go to tbs for some sort of certification process.  By that assumption it would be safe to assume that any existing mach2 vtx's will most likely not get updated firmware that is spec'd to tbs sa.

If I have to compile 3.3 (after final release) with the changes in the code I would much rather prefer to have the option to just a simple cli command (1 or 0) to only run the three different lines of code in the vtx_smartaudio files.

The thing is I don't know where I would need to put in a cli command or even how to define it without breaking anything else.

I understand simple if/else statements and how to insert those but not sure how many files I would have to alter just for a single cli command.

Thanks for your response.  No worries if you don't.

mikeller, 2018-02-09:
@ijustwannafly: I understand your pain. Unfortunately, this case is not the only one, and there is literally dozens of cases out there of 'bad clone' hardware. Mostly they are boards that have 'just one pin wrong'. If we wanted to support them all we'd very quickly be sinking into a sump of supporting 'almost working' hardware, without getting anything out of it, and leaving the users in a maze of having to pick what options to choose for their hardware.