betaflight/betaflight #5087: BF 3.3.0 - burst DMA for F3 breaks LEDStrip

TheIsotopes, 2018-01-31:
using SPRacing F3 Evo and found broken ledstrip support ... i searched and found the offender

@DieHertz #4852 and the associated build `#469` breaks led strip support on my f3evo ... pls can you check and fix it

DieHertz, 2018-01-31:
Offender? Are you kidding?
Show us `resource list` output

TheIsotopes, 2018-01-31:
Come back down I'm not joking
```
# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT
A05: ADC_CURR
A06: FREE
A07: FREE
A08: LED_STRIP
A09: FREE
A10: FREE
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: SERIAL_RX 2
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: SDCARD_DETECT
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: TIMUP 15
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FRE
```E

DieHertz, 2018-01-31:
Use PA6 or PB1 for your LEDSTRIP, or change DSHOT for PWM-based protocol

TheIsotopes, 2018-01-31:
ok i will change the cable from `ir+` to `m6` output ... then redirect resource and retesting

TheIsotopes, 2018-01-31:
@DieHertz ok ... now it's working again ... thx 👍 

PS: should possibly mention in the wiki

DieHertz, 2018-01-31:
Wiki needs editing, thanks. You could do that yourself btw, it'll be appreciated.

Please note it currently says that to use DSHOT on SPRF3EVO you have to remap motor 3 to 5, thus it working for you without remapping is a miracle already :-D

TheIsotopes, 2018-01-31:
you are right ... but i have only tested on the bench ... fc is all complete but without esc and motor

TheIsotopes, 2018-01-31:
edited dshot page on wiki to solve this problem

TheIsotopes, 2018-01-31:
must be reopen with same problem ... stand next to me today 8(

@DieHertz now i remap motor 4 to 5 and letstrip to m8 ... now again ledstrip shows wrong colors and flickers
```
# Betaflight / SPRACINGF3EVO (SPEV) 3.3.0 Jan 31 2018 / 14:27:57 (b4d399eb3) MSP API: 1.37
name SPEV
resource MOTOR 4 A06
resource MOTOR 5 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 B01
feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE
feature DYNAMIC_FILTER
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
map TAER1234
serial 1 64 115200 57600 0 115200
led 0 11,8::A:0
led 1 10,8::A:0
led 2 9,8::A:0
led 3 8,8::A:0
led 4 7,8::A:0
led 5 6,8::A:0
led 6 5,8::A:0
led 7 4,8::A:0
led 8 9,2::CW:6
led 9 9,3::CW:6
led 10 9,4::CW:6
led 11 9,5::CW:6
led 12 6,5::CW:6
led 13 6,4::CW:6
led 14 6,3::CW:6
led 15 6,2::CW:6
aux 0 0 0 1600 2000 0
aux 1 13 1 1000 1200 0
aux 2 15 0 1000 1200 0
aux 3 26 0 1600 1800 0
aux 4 36 1 1800 2000 0
set align_gyro = CW90
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set align_acc = CW90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rc_interp_ch = RPYT
set max_aux_channels = 2
set serialrx_provider = SUMD
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT150
set current_meter = ADC
set ibata_scale = 235
set pid_process_denom = 1
set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 100
set p_roll = 44
set roll_rc_rate = 145
set pitch_rc_rate = 145
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set roll_srate = 75
set pitch_srate = 75

# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: FREE
A04: ADC_BATT
A05: ADC_CURR
A06: MOTOR 4
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: SERIAL_RX 2
B00: FREE
B01: LED_STRIP
B02: FREE
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: FREE
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

# dma list
DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: TIMUP 2
DMA1 Channel 3: LED_STRIP
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

jflyper, 2018-01-31:
Try PWM4 @ A3 for motor 4.

Btw, you don't have to do this by try and error.
MCU's DMA request mapping table together with target timer assignment file would make it a easy job.
In SPRF3EVO's case, these are "STM32F303's Reference Manual RM0316" and https://github.com/betaflight/betaflight/blob/master/src/main/target/SPRACINGF3EVO/target.c

DieHertz, 2018-02-01:
Why did you remap motors? You didn't have to do that

TheIsotopes, 2018-02-01:
ok, I understood ... all right ... was my mistake

closed 2018-02-01 by TheIsotopes