betaflight/betaflight #5088: If RSSI Chennel is set to Disabled when using S.Bus then generate RSSI signal using frame drop flags from the rx

SteveCEvans, 2018-01-31:
1.) Describe your problem;
If using S.Bus receiver there is a dropped frame flag which is an indicator of signal quality rather than strength. RSSI outputs from receivers report signal strength (Received Signal Strength Indicator), but it's quality that really matters.

Betaflight currently doesn't support selection of this flag as the source of RSSI information to send to an OSD.
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
OpenPilot Revolution using S.Bus rx connected to main port (UART1) and MinimOSD connected to RxPort (UART6).
4.) Add the used configuration and firmware version;
f45c503738d8a31adbd276ea4566c81473f93bd8

[BTFL_backup_SteveBench_20180131_023259.json.zip](https://github.com/betaflight/betaflight/files/1683379/BTFL_backup_SteveBench_20180131_023259.json.zip)

5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
```# diff

# version
# Betaflight / REVO (REVO) 3.3.0 Jan 31 2018 / 20:57:49 (f45c50373) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set max_check = 2000
set fpv_mix_degrees = 15
set serialrx_provider = SBUS
set min_throttle = 1020
set gps_provider = UBLOX
set gps_auto_baud = ON
set deadband = 2

# profile
profile 0

# rateprofile
rateprofile 0
```
6.) Remove this Text :).

And here's the fix...

```
diff --git a/src/main/rx/rx.c b/src/main/rx/rx.c
index dceef40..b5bbe76 100644
--- a/src/main/rx/rx.c
+++ b/src/main/rx/rx.c
@@ -126,6 +126,9 @@ static uint8_t rcSampleIndex = 0;
 #define BINDPLUG_PIN NONE
 #endif
 
+#define RSSI_SAMPLE_COUNT 16
+#define RSSI_MAX_VALUE 1023
+
 PG_REGISTER_WITH_RESET_FN(rxConfig_t, rxConfig, PG_RX_CONFIG, 1);
 void pgResetFn_rxConfig(rxConfig_t *rxConfig)
 {
@@ -449,6 +452,14 @@ bool rxUpdateCheck(timeUs_t currentTimeUs, timeDelta_t currentDeltaTime)
             rxIsInFailsafeMode = (frameStatus & RX_FRAME_FAILSAFE) != 0;
             rxSignalReceived = !rxIsInFailsafeMode;
             needRxSignalBefore = currentTimeUs + needRxSignalMaxDelayUs;
+
+            if (frameStatus & RX_FRAME_DROPPED) {
+                   // No (0%) signal
+                   setRssiUnfiltered(0, RSSI_SOURCE_NONE);
+            } else {
+                   // Valid (100%) signal
+                   setRssiUnfiltered(RSSI_MAX_VALUE, RSSI_SOURCE_NONE);
+            }
        } else if (cmpTimeUs(currentTimeUs, rxNextUpdateAtUs) > 0) {
             rxDataProcessingRequired = true;
         }
diff --git a/src/main/rx/rx.h b/src/main/rx/rx.h
index 99c7d71..3512b18 100644
--- a/src/main/rx/rx.h
+++ b/src/main/rx/rx.h
@@ -45,6 +45,7 @@ typedef enum {
     RX_FRAME_COMPLETE = (1 << 0),
     RX_FRAME_FAILSAFE = (1 << 1),
     RX_FRAME_PROCESSING_REQUIRED = (1 << 2),
+    RX_FRAME_DROPPED = (1 << 3)
 } rxFrameState_e;
 
 typedef enum {
diff --git a/src/main/rx/sbus_channels.c b/src/main/rx/sbus_channels.c
index 4bc62cf..38548ba 100644
--- a/src/main/rx/sbus_channels.c
+++ b/src/main/rx/sbus_channels.c
@@ -69,7 +69,10 @@ uint8_t sbusChannelsDecode(rxRuntimeConfig_t *rxRuntimeConfig, const sbusChannel
     }
 
     if (channels->flags & SBUS_FLAG_SIGNAL_LOSS) {
+           // The received data is a repeat of the last valid data so can be considered complete.
+           return RX_FRAME_COMPLETE | RX_FRAME_DROPPED;
     }
+
     if (channels->flags & SBUS_FLAG_FAILSAFE_ACTIVE) {
         // internal failsafe enabled and rx failsafe flag set
         // RX *should* still be sending valid channel data, so use it.
```

jflyper, 2018-01-31:
@SteveCEvans 
IIRC, this has been requested and discussed multiple times here, and they were all withdrawn at the end for the same reason.

The frame drop counts are not signal strength indicator. Since it is a outcome of digital signal processing that probably include FECs, it does not reflect the signal strength linearly, but increases rapidly at the edge of the reception range.

I see it in two ways.

1. Frame drop count should not be confused with RSSI, and therefore treated as something totally different feature if even implemented.
2. Behavior of the count should be investigated before it is used even as an approximation of RSSI.

I'd happy to create a custom firmware  for you to do the investigation, that puts frame drops as debug value so you can observe it in BFC or OSD,

SteveCEvans, 2018-01-31:
@jflyper I implemented this on OpenPilot and found it very useful to avoid flying out of range. This implementation has been tested by putting my tx in my microwave oven and slowly shutting the door; I've yet to fly this. In practise, in flight, the OpenPilot implementation gave me plenty of warning of signal dropout. As the actual distance that this occurs at varies day to day due to humidity, it's an invaluable aid in my opinion.

And yes, you're absolutely correct, this isn't strictly speaking, RSSI. It's signal quality. For OpenPilot I modified the MinimOSD code to display RSSI and/or signal quality, but in reality when you want to know "how good is the signal?" you use RSSI if you have it, but signal quality in preference. And yes, I've modified the OSD code myself. :)

For most users, just being able to plug in their SBus rx and get an signal quality indication on their OSD without having to think about it is very useful. The only setting that needs to be changed is to set **RSSI Maximum** to 1022 in the OSD config or the RSSI indication will stick at 99, not 100%. Perhaps it would be better to have the FC send 1024, not 1023?

PR #5090 (2018-03-21): If RSSI Channel is set to Disabled when using S.Bus then generate RSS…
…I signal using frame drop flags from the rx

Fixes #5088
closed 2018-03-21 by mikeller