betaflight/betaflight #5093: Betaflight F4 Current Sensor Issue

SupaCoop, 2018-02-01:
Hi, I’m having some trouble with my betaflight F4 board. I’m running bf version 3.2.2 and config version 10.1.0. My biggest issue is current sensing. I enabled the current sensor by selecting on-board adc in the power and battery tab and I cannot get any reading from it whatsoever. Every other flight it will say -1 mah in the post flight report and 0 in the amps pulled. I tried messing with the scale and the y-intercept (can't remember what it's called, but I'm pretty sure it's just the equation for a line) but can't get the mah reading to go above 1 mah and the amperage reading to go above half an amp. The second issue I’m having is with smart audio. I enabled soft serial and assigned it to B09 in the CLI for tx1 and selected tbs smart audio in the ports tab under peripherals. But nt the osd it just shows -|-|0.  Any help and insight would be much appreciated. I’ve already contacted fpvmodel and getfpv (where I purchased it from) and they sent me over here! Hope you guys can help! 

Thanks, Cooper

jflyper, 2018-02-01:
Your issue looks like a hardware, build or a configuration problem. Please don't raise an issue here.

Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2018-02-01 by jflyper

naginni, 2018-09-18:
@SupaCoop did you fix this problem?

SupaCoop, 2018-09-18:
No I talked to fpvmodel and they said I did everything right and just deemed it defective.