betaflight/betaflight #5100: Build betaflight 3.2.4 for target KISSFCV2F7 is broken

Tschingis, 2018-02-01:
The build betaflight 3.2.4 for target KISSFCV2F7 is broken. If being flashed, there is no communication possible with the flightcontroller via bfc. 

Reproduction:

- download betaflight 3.2.4 hex for kissfcv2
- download and install kiss gui 2.0.0
- connect kissfcv2 to usb (and install required cp2102 driver if necessary)
- press RST button on kissfcv2 (force enter bootloader), the blue leds starts fast flashing
- start kiss gui 2.0.0
- select appropriate com port and connect
- click "SELECT LOCAL FIRMWARE"
- choose previously downloaded betaflight 3.2.4 hex
- flash fw
- when success, disconnect kissfc
- close kiss gui
- open bfc
- reconnect kissfc
- select appropriate com port and connect
- nothing happens , bfc reports that there was no response and closes serial port

Fix:
I can get it working by building the current master for target KISSFCV2F7. If i repeat the above process with this image bfc can connect and shows a nice 3.3.0 build. I also build the 3.2.4 tag but this didn't fix the problem, so its something in the code.

mikeller, 2018-02-01:
Sorry, we cannot support firmware that has been flashed with third party tools - there is no way for us to tell if the firmware hex file has even flashed into the correct location, or even been flashed at all.

Can you please contact the supplier to find out how to put your board into the STM boot loader mode, and then flash the firmware with the Betaflight configurator, with 'full chip erase' selected, and then try again?

Tschingis, 2018-02-02:
Ok to get things right:
BF3.3 master build -> Flashing using KissGUI -> works
BF3.2.4 build -> Flashing using KissGUI -> does not work
some prerelease BF3.2 build for that target i got from facebook -> Flashing using KissGUI -> works

Conclusion: KissGUI does not flash correctly or flash at all.

I think you might also have seen this immediatly closed ticket https://github.com/betaflight/betaflight-configurator/issues/903 there i reported that kissfcv2 cannot, and will not, be flashable with bfc because the internal bootloader is not wired against usb port. 

BF3.3 build ist working for me, so i only thought it might be useful to know that 3.2.4 does not work. Maybe someone else with more knowledge can confirm it, or not, or leave it as it is.

mikeller, 2018-02-09:
@Tschingis: As I stated before, supporting firmware that requires a closed source third party tool and boot loader to be installed and run is outside of what Betaflight can do. If the hardware is designed in such a way that flashing through the built in STM boot loader is impossible, then this board can not be properly supported by Betaflight, and KISS are the ones to ask for support.

closed 2018-02-09 by mikeller