betaflight/betaflight #5101: Minimum PWM signal for 3D mode

I-Fly-Anything, 2018-02-01:
I am was setting up I am trying to set up a Stingray (http://www.curtisyoungblood.com/legacy-product-support-curtis-youngblood/attachment/stingray-500/) with betaflight. The stingray is a variable pitch quad. It essentially has four helicopter tail rotors and a single motor. It uses servos to change the pitch of the blades for control. I am trying to adjust the endpoints. In the configurator I change Minimum Throttle on the configuration tab, but when I attempt to fly the servo still attempts to travel too far. I can adjust the travel on the high end by changing Maximum Throttle. 
```diff
# Betaflight / RMDO (RMDO) 3.2.0 Jan 31 2018 / 14:31:12 (4499933) MSP API: 1.36

feature -RX_PPM
feature -TRANSPONDER
feature RX_SERIAL
feature 3D
serial 2 64 115200 57600 0 115200
aux 0 0 1 900 1200
set max_check = 1950
set serialrx_provider = SBUS
set blackbox_p_ratio = 16
set min_throttle = 1400
set max_throttle = 1600
set min_command = 1200
set use_unsynced_pwm = ON
set motor_pwm_protocol = OFF
set motor_pwm_rate = 400
set 3d_deadband_low = 1490
set 3d_deadband_high = 1510
set 3d_neutral = 1500
set deadband = 4
set yaw_deadband = 4
profile 2

rateprofile 0
```
This is the diff from my current setup. In positive throttle, the servo barely moves (1510 - 1600). This is expected. In negative throttle, the servo still has full travel, not the short travel as seen in positive throttle. This setup uses servos to more easily see the PWM output.

I-Fly-Anything, 2018-02-01:
After digging in the mixer.c file, it looks like this is an issue when feature 3D is enabled and not DSHOT. I created pull request #5102. This should allow the flight controller to use the value input in the configuration tab for min throttle.

mikeller, 2018-02-01:
@I-Fly-Anything: I can see where you are coming from. But what you are proposing fixes the problem for your very particular situation, but breaks 3D for everybody else, by stopping them from get maximum motor output when inverted.

I-Fly-Anything, 2018-02-02:
@mikeller: This is not only a fix for me. This allows the minimum output to be controlled by the Minimum Throttle parameter in the configurator. In the current software the minimum output cannot be adjusted.

mikeller, 2018-02-02:
@I-Fly-Anything:
- `motorOutputLow` controls the maximum negative thrust in 3D mode, not the minimum;
- `min_throttle` is already used to configure the minimum throttle when in non 3D mode. Even if it was used in the correct place, it would force pilots switching 3D on and off to either have an incorrect setting for 3D mode, or for non-3D, or both. This is not desirable.

I-Fly-Anything, 2018-02-02:
Maybe I am missing something. As far as I can tell there are 5 points of interest on the throttle output for 3D quads. Hopefully you can see the image I uploaded.

[image]

(3) is center throttle. 
(2) and (4) are 3D deadband limits
(5) is max positive thrust
(1) is max negative thrust

Most of the PWM values for these points of interest are configurable with the configurator, except for the PWM value for maximum negative thrust. Currently, ```motorOutputLow``` is equal to ```PWM_RANGE_MIN``` and cannot be changed. With pull request #5102 the PWM value for maximum negative thrust can be changed by changing ```minthrottle``` in the configurator.

If my understanding and pull request is wrong, how do I change ```motorOutputLow``` so that maximum negative thrust does not occur at 1000us, ```PWM_RANGE_MIN```.

PR #5102 (2018-03-08): Fix for minthrottle when feature 3D and PWM enabled
Fixes #5101
closed 2018-03-08 by mikeller