betaflight/betaflight #5103: More crossfire telemetry data on OSD

quadflyingmedic, 2018-02-01:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

quadflyingmedic, 2018-02-01:
Would it be possible to get RSSI, LQ, and transmitter power to show on the OSD?

peteoz, 2018-02-02:
and if this is being considered, I would add to the request RFMD as well.

peteoz, 2018-03-08:
if CRSF LQ is implemented on a scale 0-300 then RFMD is no longer required on OSD

SpeedSloth, 2018-04-16:
This would be great, I'd personally like to see LQ and RFMD to know exactly what's going on or perhaps an RFMD warning when you drop down a level.

tavdog, 2022-08-28:
Close bump

haslinghuis, 2022-08-29:
Fixed in https://github.com/betaflight/betaflight/pull/8247

closed 2022-08-29 by haslinghuis