betaflight/betaflight #5105: Furious FPV RaceWhoop FC Passthrough not working

chargerdude70, 2018-02-02:
1.) Passthrough does not work if FC is updated to most current BF version (3.2.4)  

2.) Flash Furious FPV RaceWhoop to latest BF (3.2.4) and try to update esc's. When failed attempt, Flash FC back to 3.2.0 from [here](https://furiousfpv.com/download.php) to use passthrough. then flash back to desired BF FW

3. Furious FPV RaceWhoop FC and Racerstar 6a 4in1 esc 16x16mm mounting [here.](https://www.banggood.com/16x16mm-Racerstar-Star6-6A-Blheli_S-1-2S-Dshot600-Ready-4-In-1-ESC-for-FPV-Racing-p-1160389.html?rmmds=myorder)

4.) 3.2.0 RaceWhoop hex from [here]((https://furiousfpv.com/download.php)) and Standalone config 10.1.0

```# Betaflight / RACEWHOOP 3.2.0 Feb 12 2017 / 23:49:51 (1fe76a9)
name CHARGERDUDE70

feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
serial 1 32 115200 57600 0 115200
aux 0 0 1 1700 2100
aux 1 1 0 900 1300
aux 2 2 0 1300 1700
aux 3 26 2 1700 2100
aux 4 12 2 1700 2100
set motor_pwm_protocol = DSHOT300
set sbus_inversion = ON
set vbat_min_cell_voltage = 37
set vbat_warning_cell_voltage = 38
set align_gyro = CW270
set yaw_motor_direction = -1
profile 0

rateprofile 0
rateprofile 0

# ```

![3 2 0 passthrough](https://user-images.githubusercontent.com/6600288/35712731-9621aaca-0791-11e8-8c5d-77be9b07f243.PNG)
![3 2 4 passthrough](https://user-images.githubusercontent.com/6600288/35712736-9e6dbf3e-0791-11e8-8fbe-ad882fc1beca.PNG)

jflyper, 2018-02-02:
@MJ666 Do you have any info?

MJ666, 2018-02-02:
Racewhoop should be flashed with the FF_Kombini target. The 3.2 furmware referenced is an early FuriousFPV private build. Beside this I can not see a specific reason why this does not work. I don't have this hardware to test.

mikeller, 2018-02-10:
@chargerdude70: Has using the FF_KOMBINI target solved this for you?

chargerdude70, 2018-02-12:
@mikeller No it did not. I am at FF_KOMBINI 3.2.4 and still no pass-through. I had to flash a FFPV supplied target in order to update esc's. I wont bother updating esc's again but still curious as to why BF wont touch the esc's?

mikeller, 2018-02-12:
If the FuriousFPV proprietary build works, and vanilla Betaflight doesn't, it stands to reason that there is something that is specific to this board that we don't know about. The only thing you can do about this is to contact them and ask to release the source for their proprietary build back to Betaflight, as they are required by the GPL.

MJ666, 2018-02-13:
For my Racewhoop the FF_KOMBINI target and BF 3.2.5 is working with the communicating to the ESC's.