betaflight/betaflight #5107: Self pid tuning

Bugg69s, 2018-02-02:
I have an idea
Would be possible to make quad slightly change pids while flying?
Like if you punch and it will vibrate it will change the pids so it doesn't vibrate any more. 
It would be awesome

jirif, 2018-02-02:
Hi, its already feature od Betaflight, have look here https://www.youtube.com/watch?v=JpnCMLmxme0

Bugg69s, 2018-02-02:
@jirif I thought that it will adjust it self while flying on it's own

jflyper, 2018-02-03:
The correct term is **_autotune_** and **_optune_**.

phillipsj70, 2018-02-03:
I think your asking about Antigravity.

phillipsj70, 2018-02-03:
https://youtu.be/SmSWZFjXBGM

varkokonyi, 2018-02-04:
Several versions of auto-tuning mechanisms were tried, and none of them are in BF today. Guess why

jflyper, 2018-02-04:
Optune scheme was nice, together with the calculator. I was anticipating the calculator would go onboard, until the stupid dude exterminated the entire OP civilization.

ResinSmoker, 2018-02-27:
Autotune works on other platforms such as dRonin, if your board is supported.

DEvil0000, 2018-11-14:
All the autotune algorithms I have seen till now try to tune the copter from start to end at once. Thats a quite crazy and error prone approach.
I think there are more clever ways of doing this. Instead of such a all at once procedure the copter could start with the PIDs in the profile (e.g. defaults) and tune them just a bit while normal flying. The computational effort gets less critical when doing it over time and when using modern F4/F7 hardware.

Some more details about this idea:
There is a need to analyse (or aducated guess) which moments of the flight should get analysed and which not. Based on expected results calculated and real results. For example preferably propwash, sharp turns and not overdoing it on windy days.
The copter could perform algorithms similar to the ones used to render plasmatree graphs, calculate back better values and do small adjustments in the direction suggested. This way the PIDs should not jump and should improve slowly to optimal ones.

Its just a idea atm but code may proof it in the future. I would love to test code this approach but sadly I do not have much free time atm for such a time consuming project. Sharing the idea however might help someone else having time to code.

ResinSmoker, 2018-11-14:
> Hi, its already feature od Betaflight, have look here https://www.youtube.com/watch?v=JpnCMLmxme0

He was thinking more along the lines of a self-adjusting system.

DEvil0000, 2018-11-14:
> > Hi, its already feature od Betaflight, have look here https://www.youtube.com/watch?v=JpnCMLmxme0
> 
> He was thinking more along the lines of a self-adjusting system.

You can read it as both. the title says "self" and the text suggests more automatic for me.

varkokonyi, 2018-11-14:
@DEvil0000 Unless you write some code that tunes better than a trained chimpanzee, it's pointless. I want world peace for example, but simply asking for it achieves nothing

tavdog, 2022-07-25:
Close?

buzzcola3, 2022-07-25:
Yep

closed 2022-07-25 by buzzcola3