betaflight/betaflight #5112: altitude hold

DietmarM, 2018-02-03:
I do work on the altitude hold function using a racer with additional Barometer MS5611

The main problem I have is that the copter in altitude hold slightly lowers over time - starting at 1.5m until within ~20s the copter is down to the ground.

Seems baro works fine in BB and shows nice the movement down - acc calibrated several times....

reading the Code (altitude.c) I wonder about the D term:
// D 
203     result -= constrain(currentPidProfile->pid[PID_VEL].D * (accZ_tmp + accZ_old) / 512, -150, 150); 

I checked the code back to baseflight and it seems unchanged all the time.

But: should it not be accZ_tmp - accZ_old ?
D should lower a to fast changing value - adding the old and new acc does not check the Change?
What do I not understand?

DietmarM, 2018-02-04:
I think I understand it now: As it is a velocity Controller - acceleration is allready the Change. Adding the current and the previous value is may be a Kind of filter ......

closed 2018-02-04 by DietmarM

dave-patterson-bristol, 2018-02-09:
@DietmarM I am also trying to understand altitude hold with sonar/lidar/barometer. I don't get a stable hover but I do see accurate sensor readings. (Bluetooth MSP is very helpful for testing.) All the feedback on the developer groups is that the code hasn't had much attention so it's not reliable. I'm trying to fix that.

DietmarM, 2018-02-09:
Dave, same I do. What does not stable hover meens in your case?

I got it almost stable now - most disapointing for me is that sometimes the Speed Controller sees a vertical Speed due to verry small Offset in acc in z - achsis. 

I added in my case a variable to store when activating altHold the current vel. This i substract from the alt PID equation. This prevents the jump when activating at least.

to compensate the vertical drift fully (baro value gets lower by 1m without correction) I want tomorrow try to add alt I term in the PID. this should be the most effective way to adress the velocity error.

Can we get in touch with email some how through github?

dave-patterson-bristol, 2018-02-09:
I'm on the betaflight slack group. I can ask about setting up a dedicated topic for alt hold.