betaflight/betaflight #5113: Betaflight v3.3 program SP Racing F3 HEX can not be connected to the remote control

E120103016, 2018-02-03:
Today I used the Betaflight v3.3 program SP Racing F3 HEX to download. After the Configurator was set up, the RC remote control was found unable to connect. Do not know where it is.

jflyper, 2018-02-04:
> Thanks for the trouble!

You are welcome!!!

jflyper, 2018-02-04:
Your issue looks like a hardware fault or a configuration problem. Please don't raise an issue here.

Instead, please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2018-02-04 by jflyper

E120103016, 2018-02-04:
But Betaflight v3.2.4 program. Can be used without problems. For Betaflight v3.3 not. I do not know the reason. So I want to ask for help Thank you!

mikeller, 2018-02-09:
@E120103016: Betaflight 3.3 is as of yet unreleased, untested and potentially unstable. If you are not confident about setting things up, then it is advisable to stick to a released version.