betaflight/betaflight #5114: Disarm motors regardless of throttle value doesn't stick - Build #579 -

0crap, 2018-02-03:
On the latest build (#579) it's not possible to set the slider active for "Disarm motors regardless of throttle value doesn't stick".
For me that is a security risk I'm not willing to take.
Yes, I have ARM configured in Modes tab via AUX channel.

It does work as expected on the latest stable. (3.2.4 FF_PIKOBLX 11-01-2018)
With the latest stable, doing a dump, "set disarm_kill_switch = ON" is available on the CLI.
On build #579 I can't find this CLI command. Maybe this is pointing to the root cause?

AndersHoglund, 2018-02-03:
This is an intentional change, PR #5073.

wind0r, 2018-02-03:
That feature simply doenst exit anymore since it was a security risk to allow users to disable that
"Disarm motors regardless of throttle value" is on by default and cant be turned of.

The "Problem" that exit here is that your Configurator isnt compatible with the nightly firmware.
When https://github.com/betaflight/betaflight-configurator/pull/899 get merged the toggle will be hidden for new firmwares :)

0crap, 2018-02-03:
Thx for the clarification!

wind0r, 2018-02-03:
@0crap  just click the close issue on the left side from the green comment button ;)

closed 2018-02-03 by 0crap