betaflight/betaflight #5119: Mate F722 no SD Card enabled

BavariaAirview, 2018-02-04:
Hi, 

have a Matek F722 board here. 
Flashed BF3.3 and tryed to get SD logs for Tuning. 
But SD Card does not Work. 
In resource List I found SD Card not set. 

Don´t was able to get it to work. 
SD Demon was off - set it to on...   
But Resources are not set. 

In BF 3.2.4 it was set to C01.  On BF 3.3 C01 is free. 

How I have to set resource to get SD to work? 

# Betaflight / MATEKF722 (MKF7) 3.3.0 Feb  1 2018 / 15:11:24 (8738e7d) MSP API: 1.37

# name

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE
feature ESC_SENSOR
feature DYNAMIC_FILTER
set sdcard_dma= ON

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 6,6::C:3
led 1 6,7::CB:3
led 2 6,8::CB:3
led 3 6,9::CB:3
led 4 6,10::C:3
led 5 6,11::C:3
led 6 7,11::C:3
led 7 7,10::C:3
led 8 7,9::CB:3
led 9 7,8::CB:3
led 10 7,7::CB:3
led 11 7,6::C:3
led 12 5,5::AW:0
led 13 5,4::AW:0
led 14 5,3::AW:0
led 15 5,2::L:0
led 16 5,1::L:0
led 17 8,5::AW:0
led 18 8,4::AW:0
led 19 8,3::AW:0
led 20 8,2::L:0
led 21 8,1::L:0

# color

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1400 2100 0
aux 1 13 1 1800 2100 0
aux 2 26 0 1400 2100 0
aux 3 39 0 900 1200 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set moron_threshold = 128
set gyro_use_32khz = ON
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1020
set rssi_channel = 7
set rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 14
set min_throttle = 1050
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 32000
set failsafe_kill_switch = ON
set bat_capacity = 1300
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set current_meter = ESC
set vbat_scale = 109
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_rssi_alarm = 35
set osd_cap_alarm = 1050
set osd_vbat_pos = 418
set osd_rssi_pos = 2489
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2050
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2520
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2504
set osd_mah_drawn_pos = 2498
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 2445
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2187
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2073
set osd_esc_rpm_pos = 83
set debug_mode = FFT
set camera_control_key_delay = 350

# profile
profile 0

BavariaAirview, 2018-02-04:
Not even listed in feature listed: 

# feature list
Available: RX_PPM INFLIGHT_ACC_CAL RX_SERIAL MOTOR_STOP SERVO_TILT SOFTSERIAL GPS RANGEFINDER TELEMETRY 3D RX_PARALLEL_PWM RX_MSP RSSI_ADC LED_STRIP DISPLAY OSD CHANNEL_FORWARDING TRANSPONDER AIRMODE RX_SPI SOFTSPI ESC_SENSOR ANTI_GRAVITY DYNAMIC_FILTER

#

TheIsotopes, 2018-02-04:
on MatekF405 same problem ... see last four entries in #5057 ... SDCARD_CS not listed on C01 in resource list

blckmn, 2018-02-05:
Solved in https://github.com/betaflight/betaflight/pull/5121

closed 2018-02-05 by blckmn