betaflight/betaflight #5120: SmartAudio over soft serial not working

robertlacroix, 2018-02-04:
On a `LUXF4OSD` board I'm trying to get SmartAudio over soft serial working, since I don't have enough hardware UARTs available. It works fine on hardware UART 6 (TX6), but doesn't on soft serial TX (neither mapped to pin C07 - RX6, nor A01 - LED_STRIP).

Interestingly FPORT works fine on soft serial, so soft serial isn't generally broken. I'm trying to avoid it though because of the cpu overhead.

There are reports out there that say it should work on that board, but those were on 3.2, not 3.3 nightly that I'm trying.

```
# diff

# version
# Betaflight / LUXF4OSD (LUX4) 3.3.0 Feb  1 2018 / 14:22:12 (8738e7d62) MSP API: 1.37

# name

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource PWM 4 NONE
resource SONAR_TRIGGER 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 C07
resource SERIAL_RX 6 NONE
resource LED 1 NONE

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 3 1700 2100 0
aux 1 13 2 1700 2100 0
aux 2 35 1 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1024
set max_check = 2000
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set deadband = 4
set yaw_deadband = 4
set frsky_vfas_precision = 1
set vtx_channel = 7
set vtx_power = 2
set vtx_freq = 5860

# profile
profile 0

# rateprofile
rateprofile 0
```

jflyper, 2018-02-04:
You can try #5076.

robertlacroix, 2018-02-05:
@jflyper awesome thanks, #5076 fixes it!!

robertlacroix, 2018-02-09:
Fixed by #5076

closed 2018-02-09 by robertlacroix