betaflight/betaflight #5122: TIM_UP have unused DMA options

DieHertz, 2018-02-05:
As reported by someone on Facebook, LEDSTRIP has a conflict with TIM2_UP using burst DSHOT on CLRACINGF4. I've discovered that TIM2_UP (as well as TIM5 and 7) has 2 DMA stream options on both F4 and F7.

I'll submit a PR for discussion to address this issue and add `timup_dmaopt` to `DEF_TIM` macro as a temporary solution (before configurable timers become reality).

Here's a hex for the guy from Facebook to try out.
[betaflight_3.3.0_CLRACINGF4.hex.zip](https://github.com/betaflight/betaflight/files/1694338/betaflight_3.3.0_CLRACINGF4.hex.zip)

shaiss, 2018-02-25:
Can you make a version for FF_PIKOBLX?

mikeller, 2018-02-25:
@shaiss: This is not something that is universally applicable to all targets, fixes / optimisations have to be done for each target individually. You can try `dshot_burst` `on` or `off` to see if any of these fixes your problem, but apart from that you'll have to supply an exact description of the conflict that you are getting for anybody to be able to help you.

shaiss, 2018-02-26:
@mikeller.  Fair point.   I wrongly assumed that the ppl here were on the betaflight slack thread. Sorry.  I'll post the deatiks below. Board target is FF_PIKOBLX on a flying egg 130. 

@Shai Perednik uploaded a file: Screenshot_20180225-144045.png and commented: @Shai Perednik uploaded a file: Screenshot_20180225-144045.png and commented: I updated my flying egg 130 to 3m3 rc3 just now.  It fixes motor 4 not working in dshot 600. However motor 1 still needs to wired to the ppm pin and resource remapoed.  Then all motors work fine on dshot.  But as soon as I turn on the LEDs feature I loose motor 4.  Any ideas why?  Is there resource mapping issue where motor 4 and LEDs can't work at the same time?

From the screenshot below doesn't seem like it.  Full dump at https://drive.google.com/file/d/16-ouNfN2NaiVtdRD1BDfAssa5Kh9R4u8/view?usp=drivesdk

The hack I'm referring too by Albert Kim is detailed @ https://youtu.be/2TmjM5W5xmU

@diehertz & @joshuabardwell.  I've rest to defaults and turned with dshot_burst=on. all 4 motors work. soon as I enable LED Feature, no motors work and the DMA list changes completely. 

If I diable the LED feature the motors work again.  

Notes @ https://pastebin.com/hCVTztmh

Is there a way to remap the dma channels?