betaflight/betaflight #5127: No resource for SDA SCL solder pads on new DYS f4 Pro V2

jonny-questions, 2018-02-05:
[image]

'''# diff

# version
# Betaflight / OMNIBUSF4 3.1.7 May 15 2017 / 19:58:39 (e1c4b5cc2)

# name

# resources

# mixer

# servo

# servo mix

# feature
feature CURRENT_METER
feature LED_STRIP

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200

# led
led 0 6,8::C:6
led 1 7,8::C:6

# color

# mode_color

# aux
aux 0 0 0 1525 2100
aux 1 28 1 900 1425

# adjrange

# rxrange

# rxfail

# master
set 3d_deadband_throttle = 0
set motor_pwm_protocol = DSHOT600
set serialrx_provider = SBUS
set ibat_scale = 290
set ibat_offset = 20
set osd_vbat_pos = 2433
set osd_current_pos = 2401

# profile
profile 0

set d_pitch = 22
set d_roll = 20
rateprofile 0

# rateprofile
rateprofile 0

# '''

i went through the resource list and there is no pins used for SDA or SCL solder pads. I have another board of the same V2 that i flashed with 3.2.4 DYSf4PRO and there is also no pinouts on resource list for SDA/SCL solder pads. im guessing its never been updated for the new board changes

jflyper, 2018-02-06:
I2C2 resources are not assigned by default, as I2C become active as soon as pins are assigned.

closed 2018-02-06 by jflyper

jonny-questions, 2018-02-06:
Did continuity test and the sda/scl solder pads are going to b08 and b09 on the stm32 chip.

shaiss, 2018-04-26:
I'm confused, does this mean on this board we can use the uart3 tx Rx pads for i2c?  And we don't need to do anything?   If so which pad is sda and which scl.  Thank you!

ZxUkr, 2018-10-23:
So can you explain how to use I2C on this board?
Where is sda/scl located?
And what are they called?

jflyper, 2018-10-23:
@ZxUkr 
This is a board specific configuration problem, so you should contact your distributor or board manufacturer for consultation.

That being said, these two CLI commands will activate I2C2.
```
resource i2c_scl 2 b8
resource i2c_sda 2 b9
```

If this doesn't work or have trouble using an I2C sensors on those I2C labelled pads, please consider using other means of support. Thank you in advance.

ZxUkr, 2018-10-24:
> CLI commands will activate I2C2.
Maybe it is works but where are b8 and b9 pins on this board :(