betaflight/betaflight #5128: Support programming of Arduino devices in serial passthrough mode

SteveCEvans, 2018-02-06:
Support DTR in serial passthrough mode to enable programming of Arduino based devices such as MinimOSD. DTR is used to control reset on Arduino devices and must be connected in addition to the rx/tx serial lines in order to reprogram boards such as the MinimOSD.

This issue requires an additional  optional parameter to the serialpassthrough command to specify a GPIO pin to drive with DTR state passed over USB. For example to program a MinimOSD connected to UART 6 on an F4 with port PC8 used for DTR (eg OpenPilot Revolution receiver port), a value of 56 (0x38) must be used, the first nibble specifies the port, and the second the pin:

1. Use 'serialpassthrough 5 57600 rxtx 56' and then use Ardino to program MinimOSD
2. Use 'serialpassthrough 5 115200' and then use MWOSD configurator to setup

jflyper, 2018-02-06:
@SteveCEvans 
This has been considered in the past, but it requires USB driver to observe line control signal for DTR, which is not trivial.

DTR pass through (feature request / reminder) #2446

ledvinap, 2018-02-06:
@jflyper : #5129 ;-)

jflyper, 2018-02-06:
Nice!!! 😍

jflyper, 2018-02-07:
@SteveCEvans You don't have to file an issue ticket if you have a PR 😉

SteveCEvans, 2018-02-07:
@jflyper Yeah, I realise that now!

mikeller, 2018-06-10:
@SteveCEvans: This has been implemented in #5129. Can you please close this?