betaflight/betaflight #5133: Installing false Firmware FW

jummpi, 2018-02-07:
I want to install the SPFRACINGF3 Firmware on my eachine e010s. 
I installed the FRSKY3. Since then i cant install any other FW. I cant flash anything. Although i have choosen the option "full chip erase."
I can download the firmware, but when i want to flash it, this message appears:
"No response from the bootloader, programming: FAILED.
Can somebody help me to fix it?
Thx!

RipperDrone, 2018-02-07:
have u tried CLI command bl (bootloader)?

jummpi, 2018-02-07:
Are you mean the bootloader mode?

jummpi, 2018-02-07:
Hat funktioniert. Vielen Dank.

RipperDrone, 2018-02-07:
You're welcome. Can u pls close this case (helps the devs to keep the log clean). I can't close it, has to be the original author or an admin... :-) G'Nacht!

closed 2018-02-08 by jummpi