betaflight/betaflight #5137: feature: Display test messages on OSD (to show why the beeper is beeping)

jj-uk, 2018-02-08:
There are now many different reasons for the beeper to be activated, e.g failed to arm, calibrate gyro, etc..

Most flight controller boards now have an OSD.

Please consider adding meaningful text to the OSD as to why the beeper is beeping, e.g. "ARM FAIL: throttle not low"

Start up beeps:
https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#why-does-my-flight-controller-blinkbeep-lots-of-times-when-powering-up-

Counting beeps is ok I guess, but if I'm wearing goggles and it's beeping over and over when armed, it would be much faster and easier to just see on the screen that the quad cant arm because, e.g. its at an angle > 90° or the throttle isn't low, or whatever.

DanNixon, 2018-02-08:
- Arming disable reasons (the reason why you have to count beeps) are already displayed via the `WARNINGS` element.
- Failsafe is displayed as an arming disable reason.
- Battery alerts are already displayed (via `WARNINGS`).

Can you give some examples of what you feel is missing (that is worth seeing in the OSD)?

jj-uk, 2018-02-08:
I don't see any warnings except "ARMED".  Maybe I need to enable this feature? (BFF3 / 3.2.4)

DanNixon, 2018-02-08:
You just need the `WARNINGS` element enabled on your OSD config.

(`ARMED` isn't actually a warning, that is displayed separately from the rest of the OSD elements regardless of how you configure it)

jj-uk, 2018-02-11:
I needed to enable warnings! Thanks.

closed 2018-02-11 by jj-uk