betaflight/betaflight #5141: [Feature Request] Add "event markers" to blackbox log

etracer65, 2018-02-09:
The general idea is to provide a mechanism for embedding event notifications in the blackbox log so that they would show up in the viewer similarly to mode changes.

The purpose is to allow various transient events to be more easily viewed in the log.  Examples might include crash recovery activating, gyro overflow detection limiting, runaway takeoff prevention activation, etc.  Currently there is no easy way to tell these events occurred other than inferring based on other blackbox data.

Obviously this would require coordination with the Blackbox Viewer to make it aware of the new entries and display them appropriately.

etracer65, 2018-02-09:
Looks like this would be relatively easy as `blackboxLogEvent()` already exists.  Maybe I'll take a look at implementing it.  Not sure about the viewer code though as I haven't played around with that at all.

JonnyPhenomenon, 2018-02-18:
Thanks for opening this request @etracer65 - I thought this was already present,(I feel like I remember seeing at least a "beep" indicator in the logs before this) until I tried looking at a log today while tuning the crash recovery thing.  I dont have a physical beeper on this board, so I was relying on the "visual beep" feature in the OSD to tell me.  I did a hard roll and the quad definitely snapped back to level, and the OSD flashed a warning, but the blackbox logs showed nothing of the sort. (aside from the gyros and motors shooting all over)

So I searched the issues thread and see someone else got to it already. :)

closed 2026-09-26 by blckmn