betaflight/betaflight #5143: [Feature Request] VTX power output on an aux switch (Smart audio / Tramp)

DaTwo, 2018-02-09:
I would love to be able to tie the vtx power output on the smart audio / tramp setups to a AUX radio channel. This is good for a few reasons. When I fly with others I am in 200mw mode as not to bleed over adjacent channels. However when by myself i fly in 500mw mode. It would be nice to fly in 200mw mode. and if the signal gets bad I have a switch to boost the output such as flying behind something or far out.  Changing the channel on a 3 position switch would also be cool (each position is a selected channel) but not as beneficial as the power boost on on a switch.

etracer65, 2018-02-11:
Duplicate (in one way or another) with:
https://github.com/betaflight/betaflight/issues/3094
https://github.com/betaflight/betaflight/issues/4334
https://github.com/betaflight/betaflight/issues/5084
https://github.com/betaflight/betaflight/issues/4072

Maybe more.

mikeller, 2018-02-11:
Duplicate of #3094.

closed 2018-02-11 by mikeller

rauar, 2020-02-16:
I just checked all of the above "duplicates". Most of them are not duplicates at all(although related to VTX settings). #3094 is the closest one from a functional perspective. However the solution is using a lua script on the transmitter side. That means more setup hassle - especially because there seems to be multiple versions of the script which needs to be updated every now and then due to compatibility issues with newer BF releases. Plus lua scripts due to their "flexible" scripting nature are much more prone to programming errors and (I assume) are not going as much through BF's QC as core code.

I very much would prefer a more direct solution by assigning a dedicated AUX channel (identically to arm/disarm). In the modes tab a new row labeled "VTX Power" where I can configure the switch level to let's say (ideally) 3 or (at least 2) power levels would be much easier to configure. Impulse RC's Wolf PDB for example supports this. The same actually applies to VTX Pit mode: A dedicated (not the arm switch) switch via AUX channel for leaving Pit Mode (again as additional row in the modes tab).

In my eyes: much simpler, faster to setup, less error-prone, no need to deal/tinker with lua scripts

etracer65, 2020-02-16:
It can already be done. See the `vtx` CLI command.

paveltar, 2020-02-20:
> It can already be done. See the `vtx` CLI command.

> It can already be done. See the `vtx` CLI command.

What exactly do you mean?

rauar, 2020-02-20:
> It can already be done. See the `vtx` CLI command.

Wondering as well. Do you mean setting "vtx_power" via CLI ? This is not related to assigning it to an aux channel. So I think it cannot be done right now.

etracer65, 2020-02-20:
Read here: https://github.com/betaflight/betaflight/issues/9294#issuecomment-566580528

rauar, 2020-02-20:
> Read here: [#9294 (comment)](https://github.com/betaflight/betaflight/issues/9294#issuecomment-566580528)

Awesome! Exactly what I'm looking for. Will try it tomorrow. Perhaps it would make sense to add the vtx command in the wiki as well so that it can be easily found: https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings

Thanks !

mikeller, 2020-02-21:
@rauar: Thanks - contributions (in code or documentation) always appreciated - probably best add a link to the existing page to the wiki - duplicated documentation just means twice the chance of it getting outdated.