betaflight/betaflight #5157: Can't arm in cold weather

flashez, 2018-02-11:
Hi,

I am experiencing a strange problem with my Omnibus F4 board.
When I plug the lipo, I have initialization beeps but it misses the 3 final beeps that tells me that I can arm.
I first thought of a gyro problem not calibrated but no (checked on BF).
The issue only appears outside with the cold weather.
Inside no worries I can arm but as soon as it is in the cold the quad won't arm.
I tried inside without any problem, then I leave my quad outside for 5mins and I can't arm, I put my quad back inside for 5mins to warm up and it works again.
Can the gyro be bothered by the cold ?
If you have any suggestion I am open to everything.

Thanks.

spatzengr, 2018-02-11:
Put it outside for 5-minutes and take your laptop out (hopefully you have a laptop).  Plug it in and fire up Betaflight.  In the CLI tab, type "Status" and hit enter.  At the bottom of the status readout, it will list all the thing that are not allowing it to arm.  Of course one will be "CLI" because you are in the CLI tab, but there maybe be some other reasons (make sure you have the radio turned on or you will get "RXLOSS" too).

You can see a sample of the display at: 
[image]

flashez, 2018-02-11:
@spatzengr Here is what I got:
[image]
[image]
You can see the arming prevention flag 11 toggled meaning:
CALIB
Number of beeps: 11
Sensor calibration is still ongoing.
Wait for sensor calibration to complete.

So this is what I was thinking this is probably due to the gyro not initializing properly, but why only when it's cold ?

mikeller, 2018-02-11:
@flashez: Looks like your gyro is damaged (they are quite sensitive to hard deceleration, aka crashing). You can potentially improve things by increasing the setting for `moron_threshold`.

flashez, 2018-02-11:
@mikeller Thanks, gonna give it a try, will post results after

flashez, 2018-02-11:
Ok it seems to work with `moron_threshold = 78` but i'm gonna try to reduce it as low as possible, not sure what it is doing when it's too high

mikeller, 2018-02-11:
@flashez:

> Ok it seems to work with moron_threshold = 78 but i'm gonna try to reduce it as low as possible, not sure what it is doing when it's too high (default was 48)

Nothing bad should happen with a higher `moron_threshold`. It was introduced to prevent users from powering up / arming their craft while holding it in their hands. But if you have to increase `moron_threshold` it is normally an indicator that your gyro has taken some damage, and it might completely fail at any stage (most likely after more crashing).

flashez, 2018-02-11:
@mikeller ok great info, well i don't remember any hard crash but could also be bad sensor since the beginning. Gonna keep it like that until it's dead.
Thanks again.

jflyper, 2018-02-11:
@flashez Btw, how cold is the "cold weather"?

flashez, 2018-02-12:
@jflyper 0-1°C (32°F)

adrianmiriuta, 2018-02-12:
@flashez  
is this a Omnibus F4 corner ?
if yes read this :
https://www.rcgroups.com/forums/showthread.php?2934256-OMNIBUS-F4-Pro-Corner-from-AIRBOT/page35#post38947338

And ask your supplier for a replacement gyro V2 !

flashez, 2018-02-12:
@adrianmiriuta Saw this but it's not this one (Omnibus F4 v2)
But still, I have this ICM20608 gyro while on another board I have mpu6000 without any problem in cold

adrianmiriuta, 2018-02-12:
@flashez 
perhaps other Omnibus boards have this problem too.
I think the gyro box is the same !

bigd942, 2018-02-13:
@mikeller is it correct to say the gyro is damaged? I have a brand new BFF4 board (legit from reliable shop) and have had to put the moron_threshold = 120 to get it to arm without the CALIB warning. It also seems sensitive to the angle even through I normally put small_angle = 180. It wouldn't arm on the maiden so knew it had never been crashed. Seems to be a know issue with the BFF4 boards (Konasty video etc).

mikeller, 2018-02-14:
@bigd942: Either damaged or out of a bad batch I'd say. There seem to be other users complaining online about gyros on recently bought flight controllers showing a lot of noise, which suggests that there might be a bad batch out there.

bigd942, 2018-02-14:
@mikeller Thanks, I'm aware moisture can effect MEM's devices so although it's possible the cold/damp could be a factor here I agree batch related is more likely, I understand Airbot make the majority of FC's so a potential issue across lots of products. Not good, I've have 2 x of the CL Racing F4 1.2 and now this and difficult to find info on hardware issues like this before a purchase. Is it possible something can be added to the target info on the wiki to highlight concerns/known issues. Doing a quick search on Git Hub for the known issues with these two boards nothing obvious comes up or is it awareness of using the Hardware Issue label, only 3 x issues come up when I filter them which does seem low.

mikeller, 2018-02-14:
I think an issue (like the one we are discussing on)  is a more appropriate place to write down this information, due to its (hopefully) transient nature. It is also searchable, so users should find it when searching GitHub for their board's name.

mikeller, 2018-02-18:
Any more updates, or should this be closed?

bigd942, 2018-02-18:
@mikeller I've had no more issues with arming the BFF4 board with moron_threshold set to 120, prob anoth 30 x packs this week. I asked Konasty via Youtube his experience and he confirmed he's been running the same boards since Sept with moron_threshold at 100 without issue so no evidence of gyro decline. For me issue closed, OP needs to update.

Lecostarius, 2018-02-21:
Hm. I can confirm that I had the exact same Observation, copter would not arm in cold weather. At that time, it was a MultiWii 8bit board (Drotek or Flyduino, dont remember). I tried to find the reason, since the copter would reliably arm when it was inside the house (that is, when it was warm), so it was definitely temperature related. I found, just like in this case, that the accelerometer stopped working at low temperature (basically, it "froze" (pun somewhat intended) to a 90 degree Position and would not move out of there, leading the arm-angle-protection (yes, even MultiWii had that) to prevent me from arming. At that time, I doubt that we can talk about a "bad Batch" of Gyros (and in my case, it was not gyro, but acc - not sure whether it is gyro in your case, I would suspect it is also the accelerometer that is misbehaving).
So, my opinion is that accelerometers - maybe because of the way they are constructed - are susceptible to freezing (losing their function) in low temperature. Not all of them do, and it does not happen always, and it is reversible (at least in my case it was). Hope that helps.

flashez, 2018-02-21:
Anyway, my board died today with -7°C temperature. Not sure what happened but it's no longer initializing when plugging in a battery or through usb.
Board not recognized (green led still)

bigd942, 2018-02-21:
From the spec sheet of the MPU-6000 and MPU-6050 the operating temperature range is -40 degree's Celsius to +105 degrees Celsius so these temps are well within this. The ICM is -40 degree's Celsius to +85 degree's Celsius.

closed 2018-02-21 by mikeller